fhessel/esp32_https_server #6: Issue when reaching the IP, [ERR] SSL_accept failed. Aborting handshake

maxdd, 2018-06-09:
Hello,
nice library, i'm really happy to finally see an easy TLS implementation for ESP32.
I was able to compile and flash succesfully you example but whenever i try to connect to https://192.168.1.132/

i receive the following
```
HTTPSServer->debug: [-->] New connection. Socket fid is:  0x1002
HTTPSServer->debug: [ERR] SSL_accept failed. Aborting handshake.
```

i've also faced a kernel panic

```
HTTPSServer->debug: [-->] New connection. Socket fid is:  0x1001
Guru Meditation Error: Core  1 panic'ed (Unhandled debug exception)
Debug exception reason: Stack canary watchpoint triggered (https443)
Core 1 register dump:
PC      : 0x4008c704  PS      : 0x00060336  A0      : 0x4008166d  A1      : 0x3ffdac40
A2      : 0x00000000  A3      : 0x00060323  A4      : 0x00060320  A5      : 0x00000001
A6      : 0x00060323  A7      : 0x00000010  A8      : 0x800f6b5c  A9      : 0x3ffdadd0
A10     : 0x00000002  A11     : 0x00000004  A12     : 0x40085e4c  A13     : 0x3ffdad10
A14     : 0x00000000  A15     : 0x3ffdac60  SAR     : 0x0000001f  EXCCAUSE: 0x00000001
EXCVADDR: 0x00000000  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0x00000000

Backtrace: 0x4008c704:0x3ffdac40 0x4008166a:0x3ffdad10 0x4008d0e7:0x3ffdad30 0x4008d6c4:0x3ffdad50 0x40083e94:0x3ffdad70 0x40083ec5:0x3ffdad90 0x40085e55:0x3ffdadb0 0x4000bef5:0x3ffdadd0 0x400f6b59:0x3ffdadf0 0x400f6c2e:0x3ffdae10 0x400f749f:0x3ffdae30 0x400f7559:0x3ffdae70 0x400f78a0:0x3ffdaea0 0x400f7a2f:0x3ffdaf20 0x400f7bbf:0x3ffdaf40 0x400ee053:0x3ffdb5b0 0x400ee570:0x3ffdb5e0 0x400ef0bc:0x3ffdb670 0x400ef165:0x3ffdb6b0 0x400ed092:0x3ffdb6e0 0x400ece43:0x3ffdb710 0x40103b75:0x3ffdb740 0x401042f9:0x3ffdb890 0x400f2cce:0x3ffdb8b0 0x400d8894:0x3ffdb8d0 0x400d8a8b:0x3ffdb8f0 0x4014c9ab:0x3ffdb910 0x400d85a9:0x3ffdb930 0x400d2840:0x3ffdb950 0x400d365e:0x3ffdb970 0x400d538d:0x3ffdb9a0

Rebooting...
```

fhessel, 2018-06-10:
> After all the stuff i've written in this issue (now deleted, mostly kernel panics) i've noticed that the stack size of your example leads to frequent crashes. I've currently set it to 4096+256 and everything feels ok
> You should really add it in the README as a potential fix for kernel panic issues.

As the README mentions, the whole project needs proper documentation, I did not have enough time for it by now, but I created issue #7 for that and I hope I will find some time.

However, the error message you mentioned in the post's first version says: `Debug exception reason: Stack canary watchpoint triggered (https443)`, and especially for this error, the example file already [mentions increasing the stack size as a solution](https://github.com/fhessel/esp32_https_server/blob/master/https_server.cpp#L208). Again, you're right, the project needs a better documentation anyway.

> I'm still seeing connection timeout though

The timeout message will occur if keep-alive is set and no request is seen by the client for a certain period (defaults to 20 seconds). Do you experience unexpected timeouts? You can configure [this constant](https://github.com/fhessel/esp32_https_server/blob/master/https/HTTPSServerConstants.hpp#L36) to change the timeout. It's planned to expose the parameters of that file via API too, at least for those that are not required at compile time (to reserve memory).

> Also how do you #undef so that the debug output is turned off?

Undef may be the wrong word, as the macro cannot be expanded if you do so. Redefining them is the way to go.  You can replace the definitions in `HTTPSServerConstants.hpp` with this:
```
#define HTTPS_DLOG(X)
#define HTTPS_DLOGHEX(X,Y)
```

> The code above is right before the server.start() call (i've also tried after) but even if it "begins" succesfully it is not working [...]

Do you think this relates to the HTTPS server? Are you able to resolve the hostname properly if you write a minimal sketch that only sets up mDNS? I do not have that much experience with the mDNS library.

maxdd, 2018-06-10:
Regarding the timeout it seems to happen when something like this occur

```
HTTPSServer->debug: [   ] Draining response buffer
HTTPSServer->debug: [-->] New connection. Socket fid is:  0x1003
HTTPSServer->debug: [ERR] SSL_accept failed. Aborting handshake.
Fail.
HTTPSServer->debug: [-->] New connection. Socket fid is:  0x1004
HTTPSServer->debug: [ERR] SSL_new failed. Aborting handshake.
Fail.
HTTPSServer->debug: [   ] There is data on the connection socket. fid= 0x1001
HTTPSServer->debug: [ERR] An SSL error occured, fid= 0x1001
HTTPSServer->debug: [zZz] Connection timeout exceeded, closing connection. fid= 0x1002
HTTPSServer->debug: [<--] Connection has been closed. fid =  0x1002
```

Basically what you see here is a succesfull connection of the browser followed immediately by some Handshake fail (i don't know what the browser is doing at this point, surely im not the one asking for more).

Regarding the mDNS i have another sketch that worked but apparently if i compile it now it doesn't, maybe something is broken on the ESPmDNS side.
I've tried this https://github.com/espressif/arduino-esp32/blob/master/libraries/ESPmDNS/examples/mDNS_Web_Server/mDNS_Web_Server.ino

fhessel, 2018-06-11:
You need to look at the different connections. In the default configuration, the server allows for up to four parallel connections to different clients (or, which may be more accurate in your case: parallel connections to the same client application).

The `fid` that is logged with most of the entries is the internal socket ID for the connection that the log entry belongs to. So in your log, we can see four parallel connection (attempts). `0x1003` and `0x1004` fail in different stages of the SSL handshake (memory limitations may be a problem here), but `0x1002` has been opened at least 20 seconds earlier and is closed now as it was not used in the mean time by its client. I cannot tell what exactly caused the error for `0x1001` though.

However, the timeout for a connection is intended behavior. It's a trade-off between improving performance by skipping handshakes and preserving resources (as many open tls connections will notably affect your memory usage).

> (i don't know what the browser is doing at this point, surely im not the one asking for more)

Real desktop web browsers tend to do a lot of performance optimization based on the assumption that they are talking to a powerful webserver that is able to handle lots of connections easily. So what usually happens when you open https://youresp would be to open two parallel connections, one to do `GET /` and one to do `GET /favicon.ico`.

In general, the server still needs some more stability for parallel handshakes, but I'm not quite sure what would be the best way to achieve it.

maxdd, 2019-03-13:
Just came in to ask if you can somehow access the esp32 through hostname rather than through IP.
I'm using the normal WIFI.hostname command but even if i add https://xxx.local/ or xxx.local:443 i can't resolve it.

squonk11, 2019-03-13:
I think for this you need to activate mdns service. See here: [https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/protocols/mdns.html?highlight=mdns](url)

fhessel, 2019-03-13:
It depends.

On iPhone and Mac, .local address resolution is usually done by mDNS, so for a `.local`-address, you need something like `MDNS.begin("xxx")`. To my knowledge, this won't work for Windows, Linux or Android out of the box. And that's something different than the hostname used for example with DHCP.

For access via real hostname, you need to set the hostname and configure a local domain. Many consumer routers combine DHCP and a local DNS server, so that devices which are registered via DHCP can be accessed using `<hostname>.<router's local domain>`. An example: AVM's FritzBoxes (quite popular in Germany, I don't know if that's the case in other countries) use "fritz.box" as local domain. So if you use "xxx" as hostname and request an DHCP IP address, your router will bind and resolve `xxx.fritz.box` to your ESP's IP address from then. As the router advertises its DNS server via DHCP, all devices in your network should then be able to access your ESP in all browsers. Even without an mDNS implementation in the browser or Avahi client installed.

maxdd, 2019-03-13:
i made it work using 

if (!MDNS.begin(HOSTNAME)) {
        debug_println("Error setting up MDNS responder!");
        while (1) {
            delay(1000);
        }
    }

the HOSTNAME was too long and with "underscores", i guess it didnt like it.
Anyway i can now access it without using *.local since the HOSTNAME doesnt contain it explicitly

fhessel, 2019-03-13:
The `.local` part is predefined for link-local connections in mDNS.

For the mDNS hostname, it's best to stick to the recommendations in 3.5 of [RFC1034](https://tools.ietf.org/html/rfc1034), which state that letters, numbers and hyphens are the smallest subset of allowed characters for each part of domain names.

fhessel, 2019-04-02:
To summarize this issue:
- **Stack Problem/Kernel Panic**: Has been solved by increasing the stack size for the server task (and [the example](https://github.com/fhessel/esp32_https_server/blob/master/examples/Async-Server/Async-Server.ino#L87) has been changed accordingly
- **Configuration of Logging**: Can be solved by changing the macro definition as shown above. The [develop branch](https://github.com/fhessel/esp32_https_server/commit/a9fffa0a07c64b3cdbe0bb34d0ed3a43b7ee5ff4) finally also supports configuration using build options
- **Timeout after 20s**: Intended behavior to save resources
- **mDNS**: You found a solution that works for you

If I didn't miss anything, there are no problems left (for now)? Then I'd close this issue for now. If you have any further questions, bugs, ... – feel free to open another.

closed 2019-04-09 by fhessel