fhessel/esp32_https_server #8: PlatformIO HTTPS Library

thorsten-l, 2018-07-05:
Hi Frank,
i did a refactoring of your superb code to use it as a library for PlatformIO.
https://github.com/thorsten-l/ESP32-HTTPS-Library
Ciao
Thorsten

fhessel, 2018-09-05:
Hi Thorsten,
thank you for doing the refactoring and sorry that it took me some time to get back to you.

I finally also managed to organize my repo in a way that it can be used as a library for the Arduino IDE. However, I'm not very familiar with PlatformIO, but having a short look at it, the [library.json file](https://github.com/thorsten-l/ESP32-HTTPS-Library/blob/master/library.json) should be the key to make the library usable in that environment? If that's the case, would you mind if I take your file and add it to my repository as well? That would allow you and other PlatformIO users to use the library directly without you having to maintain a separate repository and merging changes every time the code is modified.

Best,
Frank

closed 2018-11-05 by fhessel

thorsten-l, 2018-11-05:
Hi Frank,

perfekt! Now I can delete my repository.  :-)

Ciao
Thorsten

commit d3dd834d (2018-11-05): Added library.json for PlatformIO. Close #8