fhessel/esp32_https_server #17: HTTP support

BojanJurca, 2018-09-01:
Hello! I just finished adding (unencrypted) HTTP functionality for my home project. I need this because it works much faster than HTTPS when I connect to the ESP from my home WiFi. Although it will be difficult to keep up with your upgrades in the future. I would like to suggest that you include HTTP in your project, if it does not ruin your concept?

fhessel, 2018-09-01:
I had this in mind when creating the code, so the whole HTTPS stuff _should_ mainly be done in the `HTTPSServer` and `HTTPSConnection` class – note that many other classes just start with `HTTPsomething` and should be generic for both protocols. However, parsing the request line and header is not really separated from the SSL connection handling by now, but this can be done by refactoring the `HTTPSConnection` class so that it does no more than handling the actual SSL stuff and putting everything else in a new class. That would allow to implement an `HTTPServer` and `HTTPConnection` that both sit on top of that new class. Creating an instance of `HTTPSServer` _and_ one of `HTTPServer` with the same `ResourceNode`s being configured should then allow you to handle both protocols.
However, that's not at prio 1 for me, as there are HTTP available for the ESP32 and solving some other issues is more urgent for some personal projects I'm working on at the moment.

BojanJurca, 2018-09-01:
This is exactly what I have done. I introduced additional listening socket in HTTPSServer. Everything else is the same except for minor changes in some functions that read from and write to SSL - they now read directly from and write directly to socket and this is it. All your HTTPS functionalities work under HTTP as well.

BojanJurca, 2018-09-01:
... exept for refactoring and new classes. Le me upload the code and then you can decide whether to keep it or not.

fhessel, 2018-09-01:
Thank you for the pull request! I'll have a look at it and see how I can integrate it.

fhessel, 2018-09-02:
I did some refactoring, so now you have an `HTTPServer` class as well as an `HTTPSServer` class. You can either use one of them to serve just one protocol or create two instances and run them in parallel (see `https_server.cpp` for an example).

My main focus for the refactoring was to have an additional abstraction layer between handling HTTP, so the HTTP**S** classes now only do the stuff that's specific for TLS. I hope that's also a suitable solution for your project. Thank you again for providing the code!

closed 2018-09-02 by fhessel

BojanJurca, 2018-09-02:
Great work! Thank you. I'll go with your version of course. I just have to
make minor changes in my project to adjust to it.

BojanJurca, 2018-09-02:
Frank, I have problems compiling new code with Arduino. Beginning with:

HTTPConnection.cpp:449: error: macro "bind" passed 4 arguments, but takes
just 3

       next = std::function<void()>(std::bind((*itMw), &req, &res, next));

Any idea how to solve this?

>

fhessel, 2018-09-02:
That line is for a change I made for #10. The `bind` in that line _should_ be expanded to the `bind` declared in the functional header of the standard library, but your compiler seems to go for the socket one (which requires exactly three parameters, not one or more). At least that's how I would interpret that error.

I'll setup a small Arduino project to be able to reproduce it and then try to organize the includes in a non-conflicting way.

fhessel, 2018-09-02:
I can build the project without any problems in the Arduino IDE.

Could you provide me with some additional info? I think being able to reproduce your problem will make it easier to find its cause, rather than randomly changing some includes and let you try again and again.
Did you use the latest version of [arduino-esp32](https://github.com/espressif/arduino-esp32)? Which OS do you use? Which version of the Arduino IDE?

If you just want to have HTTP/HTTPS in parallel, without the middleware feature, you can work with [commit `c5d8a1e28c`](https://github.com/fhessel/esp32_https_server/tree/c5d8a1e28c39ddac6f090c9365bcd48a9879c81a) for now.

BojanJurca, 2018-09-02:
I use Windows 10. I'll install Arduino and ESP tools again from the scratch
and try again ...

BojanJurca, 2018-09-02:
After reinstalling Arduino it works. Thank you!

commit ea6fb5bc (2018-09-02): Refactoring of Pull-Request #18 (HTTP Support). Closes #17