fhessel/esp32_https_server #20: Instability issue

BojanJurca, 2018-09-17:
Hi Frank, I’ve spend quite some time testing HTTPS server regarding stability which I’m having problems with. What I have found out so far is that your HTTPS server works pretty much OK. I strongly suspect the reason for instability lies in LWIP stack. As such, it influences the stability of the whole solution.  As long as I am accessing HTTPS server via local WiFi everything works fine for days (this is the longest time I have tested). However, as soon as I open it to the internet and access it, for example from my mobile phone, ESP32 freezes in few hours. It seems that the poorer the connection is the sooner ESP32 freezes. I wonder if you have experienced the same problems and how did you handle them?

fhessel, 2018-09-17:
From my point of view, there are some circumstances that will (maybe combined with each other) lead to the symptoms you describe:
- First, I'm pretty sure the server isn't free of bugs, and there might even be some problems in memory handling that I'm not aware of by now, and which might cause memory exhaustion after some period of time. The HTTP implementation is really simplistic and by far doesn't cover every detail of the standard.
- Being available on the internet, especially on ports 80 or 443, the ESP32 will face much more traffic than what you intend it to process. There might be port scanners starting handshakes that are never torn down gracefully, you might receive requests from web crawlers etc. - all of which belongs more to the _unexpected_ kind of traffic. I did not test shooting many SYN packets at the server, but I won't be surprised if that causes problems, either in the lwip stack or because the server's resources are exhausted quickly (I need to limit the amount of parallel connections to a very low bound, as they might potentially lead to memory-intensive SSL handshakes – we're still running on an embedded device).
- Weak connections might worsen the problem, as missing frames in the SSL communication or connections being terminated during transmission also require the server to clean everything up.
- There seem to be some errors that I'm still not able to handle within the server, e.g. if `SSL_accept` faces memory issues. There seems to be a function somewhere down in the SSL implementation that does not check for errors during memory allocation, so that a stack canary is triggered which eventually leads to a full reboot of the ESP32. This kind of bugs would require some more analysis and then have to be filed as issues in the esp-idf repository. At least when I started writing the first parts of this lib, it seemed to me that the server part of the ESP32's SSL implementation is not as well documented or received as much attention of the community as some other modules, so maybe it's just not that much tested by now.

I did only use the library on the local network by now and found it to be quite stable, and by now, I don't have any intentions myself to allow a direct connection from the internet. To find out more about the reasons for the instability you experienced, one would need to capture the traffic that hits the ESP32 and try to identify what kind of traffic causes the freezing. That may allow to find suitable countermeasures depending of the origin of the problem. However, that will be relatively complex and time-consuming, as you need to have a setup that allows recording of the traffic as well as to merge those traffic logs to the ESP32's log.

On short-term basis, I can only think of solutions like a reverse proxy server in the network or something else that is able to understand the HTTP traffic and is able to drop malformed requests before they hit the ESP32, but as it requires additional hardware (with more computational power, if you don't want to move the problem somewhere else), that might not be applicable for every project.

So the next step to work on this would be to identify exactly what is causing the instability and then to try to fix that.

me21, 2018-11-20:
I've conducted the following experiment:

I started HTTPS server on port 443, and also DNS server like this: `dns.start(DNS_PORT, "*", WiFi.softAPIP());`. This line redirects each and every hostname lookup to the IP address of ESP32 itself.

When I create an access point and connect to it with my laptop, I can see endless flow of connect attempts from my antivirus, browser, messengers, etc. They all try to connect to their servers, but connect to ESP32 instead. They all fail handshakes because ESP32 has self-signed certificate.

Even so, it breaks something inside ESP32, so that I cannot ping it in a minute or so. It keeps running, but something weird happens with its WiFi part, so I guess it's unstable anyway.

Without such DNS redirection (and, as a consequence, without so many HTTPS connect attempts) I don't observe such behaviour.

So I guess, your instability issues can be a) reproduced, b) avoided by hiding ESP32 behind a firewall.

BojanJurca, 2018-11-20:
Hello to both of you. Thank you for your trouble. I have experimented a little with a very simple socket server working on port 80 (without https server installed) and the results were more or less the same, so the problem is not in https server itself. I guess ESP32 is just not meant to be connected directly to the internet as you both have suggested. One way or another additional peace of hardware may be necessary.

fhessel, 2019-03-26:
I'll close this for now as it's been inactive for some time and might not be directly connected to the server itself.

closed 2019-03-26 by fhessel