fhessel/esp32_https_server #23: setupCert failed

gigirizzi, 2018-11-06:
hi when  start my securserver  i have setupCert failed, i try my cert.h an private key withexample and all wark...

fhessel, 2018-11-06:
Hi,
could you please describe the process that you used to create cert.h and private_key.h? Did you use the create_cert.sh script in the extras folder?

If you created those files on your own, make sure to use RSA as algorithm, as the ESP32 needs different API calls for ECC cryptographic keys that I did not implement yet. I suggest 1024 or 2048 bits as key size, otherwise you might run into trouble due to the limited memory of the ESP32.

gigirizzi, 2018-11-06:
i have add log in httpsserver class.. the problem is in 

	ret = SSL_CTX_use_RSAPrivateKey_ASN1(
			_sslctx,
			_cert->getPKData(),
			_cert->getPKLength()
		);
here is the fault...

fhessel, 2018-11-06:
That's the part of the code that will interpret the content of your private_key.h file.

If the file has been generated correctly, it should look like this:
```c++
#ifndef PRIVATE_KEY_H_
#define PRIVATE_KEY_H_

unsigned char example_key_DER[] = {
  0x30, 0x82, 0x02, 0x5c, 0x02, 0x01, 0x00, 0x02, 0x81, 0x81, 0x00, 0xe1,
  // ... (many more lines)
};
unsigned int example_key_DER_len = 608;

#endif
```

`example_key_DER_len` has to match the amount of elements in `example_key_DER[]`.

And the private key has to be of the correct type (RSA 1024 or 2048 bit) and it has to match the certificate in your cert.h, which should look like this:
```c++
#ifndef CERT_H_
#define CERT_H_

unsigned char example_crt_DER[] = {
  0x30, 0x82, 0x02, 0x19, 0x30, 0x82, 0x01, 0x82, 0x02, 0x01,
  // ...
};
unsigned int example_crt_DER_len = 541;

#endif
```

If that `SSL_CTX_use_RSAPrivateKey_ASN1` gives you an error, one of these conditions does not hold, and you are trying to use invalid cryptographic parameters.

So if I should help you, I need to know how you created the private_key.h and cert.h file. Maybe try to run extras/create_cert.sh again and check the log for errors or post the complete create_cert.sh log here, then I'll have a look at it.

gigirizzi, 2018-11-06:
i found the problem but i don't understand... if i copy the value of private_key.h in my ino all work, in the external file don't work.....

fhessel, 2018-11-06:
Just to understand it correctly: Compiling, Linking and Uploading your script to the ESP32 is working, but you get a runtime error on that `SSL_CTX_use_RSAPrivateKey_ASN1` line. But if you do not use `#include "private_key.h"` and instead copy the definitions directly into your main sketch, you can build it, upload it and it runs without an error?

fhessel, 2018-11-08:
Do you need further assistance with this or did you get it running properly and I may close this issue?

fhessel, 2018-11-11:
I assume that's not the case. If you face any other problems, feel free to open a new issue.

closed 2018-11-11 by fhessel