fhessel/esp32_https_server #25: Template processing

me21, 2018-11-20:
Please add the template engine.

It would be nice to be able to store the HTML file in SPIFFS and replace certain placeholders in it with their real values depending on some conditions. A callback function provided by user could return these values.

A similar feature exists in ESPAsyncWebServer: https://github.com/me-no-dev/ESPAsyncWebServer#template-processing

fhessel, 2018-11-20:
Hi,
thanks for the feature request. That sounds like a useful addition to the library.

As far as I see by now, this will require #15 to be implemented first (ESPAsyncWebServer also does chunked transfer encoding in case the file contains templates).

Regarding storing the template files, I would prefer to keep SPIFFS etc. out of the server's code to minimize dependencies of the library and have a maximum of flexibility wrt. where you place your static files (SPIFFS, SD Card, Progmem, ...). By using the default handler you can add SPIFFS access relatively easy to your server and put a REST API on to. There are already examples out there on how to do this, look for instance at [this blog post by Nathan McMinn](https://nathanmcminn.com/2018/09/23/a-modern-micro-web-stack-for-the-esp32/).

I can put this onto my to do list, but I'll not be able to do the implementation by myself in the next few weeks.

dethegeek, 2020-11-15:
Hi

I have the same need and build my own template implementation. It is currently limited to placeholder and include support, but this may be be a starting point for you and other people.

I just created a repo to share it here : https://github.com/dethegeek/esp-template . I apologize for the README file : I wrote it from scratch, may be incomplete or have some errors: i did not planned to share it as a standalone library. Questions / issues / PRs are welcome.