fhessel/esp32_https_server #26: Provide instructions for creating certificate on Windows

me21, 2018-11-20:
Certificates can be created on Windows with OpenSSL like this:

First, generate CA certificate:
```
C:\OpenSSL-Win32\bin\openssl req -x509 -new -key rootCA.key -days 10000 -out rootCA.crt
```

Then create cmd script with the following content:
```
C:\OpenSSL-Win32\bin\openssl req -newkey rsa:2048 -out %1.csr -keyout %1.key -nodes -subj "/C=US/ST=GE/L=location/O=organisation/CN=%1"
if "%~2"=="" (
  set SAN=DNS:%1
) else (
  set SAN=DNS:%2
)
C:\OpenSSL-Win32\bin\openssl x509 -req -in %1.csr -CA rootCA.crt -CAkey rootCA.key -CAcreateserial -out %1.crt -days 5000 -extfile extfile.cnf
C:\OpenSSL-Win32\bin\openssl x509 -in %1.crt -out %1.crt.der -outform der
C:\OpenSSL-Win32\bin\openssl rsa -in %1.key -out %1.key.der -outform der
```

Save this snippet to `create_new_cert.cmd` file and run it as `create_new_cert elabel.local`. It will create new certificate named `elabel.local.crt` and corresponding key too. It will use `DNS:elabel.local` also for `subjectAltName` field of the certificate.
If run with two arguments (`create_new_cert filename dnsname`), it will create certificate with the corresponding filename and put `DNS:dnsname` into `subjectAltName` field of the certificate.

You may further tailor the script as you need.

me21, 2018-12-02:
There are many results when googling for "bin2c", here's one link: https://www.segger.com/free-utilities/bin2c/

flywire, 2019-07-08:
> First, generate CA certificate:

After [Create Root Key](https://gist.github.com/fntlnz/cf14feb5a46b2eda428e000157447309#create-root-key)