fhessel/esp32_https_server #29: Support forms in POST

me21, 2018-12-03:
I tried to send POST requests with forms. Tried both x-www-form-urlencoded and multipart/form-data, but request params array is empty in all cases. Does the server support handling HTML forms? Am I doing something wrong?

fhessel, 2018-12-03:
No, you're not. Request params will contain only the values passed after the question mark in the **request URL**, like https://esp32.local/?foo=bar which will lead to `bar` being assigned to parameter `foo`.

Both `x-www-form-urlencoded` and `multipart/form-data` will send the form data to the **body** of the POST request (see [these](https://developer.mozilla.org/en-US/docs/Web/HTTP/Methods/POST#Example) MDN examples for the details on the differences in data structure).

At the moment, the server library does not support any special form body parsing, as I expected it mostly to be used for REST-like services, which can use other third-party libraries like ArduinoJSON. So by now, you would have to do it on your own by reading the request body. If you use urlencoded forms, you could start by reusing my [implementation for parsing URL parameters](https://github.com/fhessel/esp32_https_server/blob/master/src/ResourceResolver.cpp#L40). Besides the initial `?` character, it should be very similar (be aware of #19 in case you're sending other things than letters or numbers).

I'll tag this as a feature requests, but as you might've seen by my reaction on other issues, I'm a bit short on time at the moment.

me21, 2018-12-03:
I totally understand. Thanks for replying.
I need to upload files to ESP32, so I guess I need multipart/form-data here. I'll try to implement request body handling myself, thanks!

me21, 2018-12-06:
I've implemented `multipart/form-data` parsing. I have no obvious bugs on a simple form with one file input and one text input.

I'm attaching my source code here for future reference as a starting point.

```
void handleNewConfig(HTTPRequest* req, HTTPResponse* res) {
  auto contentType = req->getHeader("Content-Type");
#ifdef DEBUG_MULTIPART_PARSER      
  Serial.print("Content type: ");
  Serial.println(contentType.c_str());
#endif
  auto boundaryIndex = contentType.find("boundary=");
  if(boundaryIndex != std::string::npos) {
    //TODO: remove all magic constants
    auto boundary = contentType.substr(boundaryIndex + 9);
    auto commaIndex = boundary.find(';');
    boundary = "--" + boundary.substr(0, commaIndex);
    if(boundary.size() > 72) {
      handleAdminHtml(req, res);
      return; //TODO: error 500, RFC violation
    }
    auto lastBoundary = boundary + "--";
    // Stream the incoming request body to the response body
    // Theoretically, this should work for every request size.
    std::string buffer;
    // What we're searching for
    enum {BOUNDARY, CONTENT_DISPOSITION, DATA_START, DATA_END} searching = BOUNDARY;
    std::string fieldName;
    std::string fieldValue;
    bool isFile;
    // HTTPReqeust::requestComplete can be used to check whether the
    // body has been parsed completely.
    while(!(req->requestComplete())) {
      const auto oldBufferSize = buffer.size();
#ifdef DEBUG_MULTIPART_PARSER      
      Serial.print("Old buffer size: ");
      Serial.println(oldBufferSize);
#endif
      buffer.resize(256);
      // HTTPRequest::readBytes provides access to the request body.
      // It requires a buffer, the max buffer length and it will return
      // the amount of bytes that have been written to the buffer.
      auto s = req->readChars(&buffer[oldBufferSize], buffer.size() - oldBufferSize);
      if(!s)
        break;
#ifdef DEBUG_MULTIPART_PARSER
      Serial.print("Reading buffer (");
      Serial.print(s);
      Serial.print(" bytes): ");
      Serial.println(buffer.c_str());
#endif
      buffer.resize(s + oldBufferSize);      
      // reading line by line
      std::istringstream is(buffer);
      for(std::string line; is.good(); ) {
        std::getline(is, line);
#ifdef DEBUG_MULTIPART_PARSER      
        Serial.print("Next line is ");
        Serial.print(line.size());
        Serial.print(" bytes: ");
        Serial.println(line.c_str());
#endif
        // remove preceding \r
        bool crRemoved = false;
        if(!line.empty() && (line.back() == '\r')) {
          crRemoved = true;
          line.pop_back();
        }
        if(is.good()) {
          switch(searching) {
[... 99 lines cut ...]
            Serial.print("Boundary found without LF: ");
            Serial.println(line.c_str());
#endif
          } else {
            // no boundary
            // to avoid the case when the buffer is full (256 bytes), we add it to field value here
            //TODO: it is possible to cause hangups by passing long strings during (searching != DATA_END),
            // but for now we assume the client is well-behaving
            if(searching == DATA_END) {
              if(isFile && crRemoved)
                line.push_back('\r');
#ifdef DEBUG_MULTIPART_PARSER      
              Serial.println("Adding it.");
#endif
              fieldValue += line;
              line.clear();
            }
          }
          buffer = line;
        }
      }
    }
  }
}
```

fhessel, 2018-12-06:
Thank you for providing the code, that will be a really good starting point!

I also thought about how one could deal with large uploads and I came up with an API that allows to iterate over the body's fields and then read each field step by step using a buffer, so that you could e.g. directly write it to an SD Card. That should allow for arbitrary body sizes, but comes with the downside of not being able to access random fields of the body. To address both forms of body encoding (as urlencoded has way less overhead for short text values) there could be a parser for each content type based on the same body parser API, making the content encoding easily interchangeable. Usage should then be something like this:

```c++
void handleRequest(HTTPRequest * req, HTTPResponse * res) {
  HTTPMultipartBodyParser * bParser = new HTTPMultipartBodyParser(req);
  while(bParser->nextField()) {
    std::string fieldName = bParser->getFieldName();
    byte buf[128];
    size_t len;
    while(bParser->getRemainingLength()>0) {
      len = bParser->read(buf,128);
      // Do something with buf[0..len] and fieldName
    }
  }
}
```

It's just a draft of the API and most likely won't even compile at the time, but I committed it for reference to the bodyparser branch (7861a98f1cebdebe623ceee85e6545c3f0757f50).

jackjansen, 2020-02-16:
I'm making good progress implementing this, based on the bodyparser branch, but I'm running into an issue with the suggested API. `getRemainingLength()` and actually `getLength()` as well aren't really implementable without first reading the whole content of the multipart part. The part header doesn't contain a field that defines the size of that part, so that means you basically have to read all of it before you can know the size. (Not a great design of the MIME multipart: it also means you have to search for CRLF followed by the separator string in binary data, brrrr). I think I'll replace the two size calls a `bool endOfField()`, unless you have a better suggestion.

fhessel, 2020-02-16:
That API was really only a suggestion on how one could implement this, and I see the problem. I'd clearly put the ability of reading parts of arbitrary length over the ability to get the length in advance, so your `endOfField()` is the better choice here.

> it also means you have to search for CRLF followed by the separator string in binary data, brrrr

I think binary data may be the easy case. If the client decides to make use of a `content-transfer-encoding` (see section 4.3 in [RFC2388](https://www.ietf.org/rfc/rfc2388.txt) in combination with section 6 in [RFC2045](https://www.ietf.org/rfc/rfc2045.txt)), it might be necessary to do an additional conversion step. I don't know how current clients do behave, so I'd expect `binary` for files and `8bit` or maybe even `quoted-printable` for text fields being the default. If the clients do switch between different variants, I'd suggest to implement the most common one first and then add converter functions with the same signature later on, maybe something like:

```c++
/**
 * Removes quoted printable content transfer encoding
 * data_in buffer to read from
 * size_in size of data_in buffer
 * data_out buffer to write the decoded data to
 * size_out number of bytes written to data_out
 * returns: Number of bytes read from data_in
 */
size_t transferDecodeQuotedPrintable(uint8_t *data_in, size_t size_in, uint8_t *data_out, size_t &size_out);
```

That should allow handling all encodings while the caller can remain agnostic of the symbol length in the encoded data. He only has to check for `<CR><LF><boudary>` in the incoming data. If `size_in > return_value`, the caller knows that part of the data couldn't be processed, most likely because only a partial input symbol is present in `data_in`.

jackjansen, 2020-02-21:
I've created a pull request. I'm not 100% happy with the code, especially because I've had to implement yet another level of buffering (I saw the  the existing buffered reading code but it's in private methods, and I've decided to implement my own because of the tricky way to test for boundaries while reading binary data). Performance seems sort-of decent despite the extra copying.