fhessel/esp32_https_server #30: Trying to receive a telegram hook

denisbrondy, 2018-12-16:
Hello and tks for this promising library !
I'm trying to run this https server on my ESP32 to get a telegram bot post on it using a web hook : https://core.telegram.org/bots/api#setwebhook

I successfully ran up to a point I don't understand what's happening. It seams the ESP get block while reading the request coming from telegram. Here is what is received on the console when telegram makes a POST on the web hook.

26004 HTTPSServer->debug: [-->] New connection. Socket fid is:  0x3A
28952 HTTPSServer->debug: [   ] There is data on the connection socket. fid= 0x3A
28954 HTTPSServer->debug: [   ] Request line finished: method=POST, resource=/telegram/webhook
28958 HTTPSServer->debug: [   ] There is data on the connection socket. fid= 0x3A
28965 HTTPSServer->debug: [   ] Header: Host:somewhere.in.france.fr
28970 HTTPSServer->debug: [   ] Header: Content-Type:application/json
28976 HTTPSServer->debug: [   ] Header: Content-Length:347
28981 HTTPSServer->debug: [   ] Header: Connection:keep-alive
28987 HTTPSServer->debug: [   ] Header: Accept-Encoding:gzip, deflate
28993 HTTPSServer->debug: [   ] Headers finished
28998 HTTPSServer->debug: [   ] Resolving resource...
29002 HTTPSServer->debug: [   ] Testing simple match on /telegram/webhook
29008 HTTPSServer->debug: [   ]   It's a match! Path:/telegram/webhook
29015 HTTPSServer->debug: [   ] Keep-Alive activated. fid= 0x3A
**29020 HTTPSServer->debug: [   ] Creating buffered response. Buffer size:  0x578**

It stop at this line and after that, it does not answer to any other request.... Could it be linked to the fact the library is not able to read the body data (JSON) which is included ? I tried to read the bytes of the req using a while loop and requestComplete() method but still the same.

I'm using a classic doit dev ESP32 board, with platformio and associated software up to date.

Any idea ?
Tks :-)
Denis

fhessel, 2018-12-17:
Did you use the latest version of the `master` branch to test this or the `v0.1.1` release? The release might already/still include issue #28 which means as soon as you try to read the body of the request, you'll get stuck in an endless loop. 4aeab6106f08b78e5bde214102936885a02778e2 fixes this in the current `master` branch. At least that would fit to the behavior you experienced with the server no longer answering any other requests.

Regarding parsing the body: By now, the server does not provide any help for actually parsing the body, besides for providing you with the raw bytes. So you'll need to do that on your own. My intention is not to re-invent the wheel and blow up the library's size with parsing of content-types that are not strictly bound to HTTP. For example if you want to use JSON, [ArduinoJSON](https://github.com/bblanchon/ArduinoJson) does a great job, and you can just forward the bytes you receive from the `HTTPRequest` to that library.

denisbrondy, 2018-12-17:
Hi Frank,
Yep, you are right ! Sorry, didn't take time to look at other issues. 
In fact, i'm using the version 0.2.0 released in platformio. I don't see this tag in git and I don't know how you manage the publishing of the library in platformio.
Anyway, everything is fine and I confirm you that I will use ArduinoJson, I planed to manage the response body this way !
Tks again,
Denis

closed 2018-12-17 by denisbrondy

fhessel, 2018-12-17:
Hi Denis,
good to know that this solves your issue.

I did not push anything to PlatformIO by myself, as I wanted to get familiar with the ecosystem before I do so, to stumble upon possible issues and not let others do that for me.
However, I already provided a library.json to allow using the library by simply cloning the repository into your project's lib folder.

In the `master` branch, the library.json is already set to 0.2.0 so that I don't forget to increase it when tagging the next release, but I didn't mean to publish that version right now.
As far as I understood, the PlatformIO library crawler should only take "version-like" tags or branches of a repo and publish them as a version (but not the master), so I'm not sure if using 0.2.0 will create stable builds for you if I update the `master` branch. So I assume, if you use 0.2.0 at the moment, you will get the most-recent master branch.

At least the PlatformIO library points to this repository, so whoever made that entry did it probably just for his/her convenience.