fhessel/esp32_https_server #36: How to connect University network Internet (https://hotspot.xx.edu/login)

williamesp2015, 2019-03-18:
Hardware:
Board:	ESP32 Dev Module?
Core Installation version:	Updated
IDE name:	Platform.io
Flash Frequency:	40Mhz
Upload Speed:	115200?
Computer OS: Windows 10 , Windows7

Description:
Describe your problem here
I connected to the university using SSID and Password given by the ICT unit. I tried many format of http.begin to connect the Internet. Manually I go to hotspot.login.html page and type username and password. The code is below:
<form name="sendin" action="https://hotspot.XX.edu/login" method="post">
  alert(username.value+"--"+password.value);
  document.write(username.value+"--"+password.value);
    <input type="hidden" name="username">
    <input type="hidden" name="password">
    <input type="hidden" name="dst" value="http://XX.edu/">
    <input type="hidden" name="popup" value="true">

 function doLogin() {
    document.sendin.username.value = $('#username').val();
    document.sendin.password.value = md5($('#passwordInput').val());
    document.sendin.submit();
    return false;
}
ICT unit recommended to set username and password by Unix:
$ curl –data “username=&password=” https://hotspot.XX.edu/login
$ curl http://XX.edu/logout
I get error using differing format like:
http.begin("https://hotspot.XX.edu/login username=&password=");
Serial output:
[HTTP] begin...
[HTTP] GET...
[E][WiFiClient.cpp:243] setSocketOption(): 1006 : 9
[HTTP] GET... code: 200
Any conflict between https and http.get() may caused this error?
Thank you

fhessel, 2019-03-18:
I don't get the way that in which this is related to this library, which provides an HTTPS _server_ library.

From what I see in your code, you try to use the HTTPClient library of the ESP32 Arduino Core to send credentials to the hotspot's login server as HTTP _client_ – which is correct.

However, if you're trying to connect to an HTTPS server, you most likely will need to setup a root certificate like in [this example](https://github.com/espressif/arduino-esp32/blob/master/libraries/HTTPClient/examples/BasicHttpsClient/BasicHttpsClient.ino) to get everything working. Besides that, I don't know the way these hotspots are working, but if they suggest do use `curl` with a `--data` parameter, you usually need a `POST` request instead of a `GET` and put the url-encoded data in its body.

If there's any relation to this library that I'm missing currently, please let me know.

williamesp2015, 2019-03-20:
Thanks fhessel .  I tried using POST instead of GET but not success in sending correct parameters. I see  in the function doLogin()  password encoded using md5 that means I have to port md5.js in my program first and then POST it by correct format :

    <form class="col-lg-12 login-form" action="https://hotspot.XX.edu/login" id="sign_in" method="post" onsubmit="return doLogin()">
                    <input type="hidden" name="dst" value="http://XX.edu/">
                    <input type="hidden" name="popup" value="true">
                    <div id="errorval" class="alert bg-red" style="display: none;">
                        <button class="close" data-close="alert"></button>
                        <span id="errorMsg"></span>

function doLogin() {
document.sendin.username.value = $('#username').val();
document.sendin.password.value = md5($('#passwordInput').val());
document.sendin.submit();
return false;
}
Any help would be greatly appreciated.

fhessel, 2019-03-20:
I still don't see the connection to the library, but:

- If the credentials do not change at runtime, compute the MD5 hash during implementation and store the hashed password. That's _way_ easier.
- The ESP-IDF (which is also the foundation of the ESP32 Arudino Core) is shipped with an implementation of [mbedtls](https://github.com/espressif/esp-idf/tree/master/components/mbedtls), that you can use to calculate md5 sums on the device. A starting point to dive into the documentation could be [this page](https://tls.mbed.org/api/group__hashing__module.html).
- Most probably, as the hash is binary data, you'll need to encode it somehow. Your code snippets don't show what your JavaScript `md5()` function is doing, but usually it will be either [URL-encoded](https://en.wikipedia.org/wiki/Percent-encoding) or [Base64-encoded](https://en.wikipedia.org/wiki/Base64). The latter is again implemented in mbedtls, and this library already uses it. You may want to have a look at [`websocketKeyResponseHash()` in `HTTPConnection`](https://github.com/fhessel/esp32_https_server/blob/master/src/HTTPConnection.cpp#L636) to see how it's done, but I'd strongly suggest to also read the API docs for `mbedtls_base64_encode`.

fhessel, 2019-03-23:
Are there any actual questions/problems with the library remaining? Otherwise I'd close the issue for now.

closed 2019-03-29 by fhessel