fhessel/esp32_https_server #37: Plaintext instead of HTML

dorarosi, 2019-03-21:
So I build a HTTPS Server wich reads files from SPIFFS. Sometimes the File is just displayed as plaintext instead of HTML

**How To Reproduce**
Use the code in next Post.
Upload some files to SPIFFS.
Try.

**Expected Behavior**
I want to display HTML Files.

**Actual Behavior**
I get the HTML code as plaintext startet with 
`Content-Type: text/html`
`Connection: close`

**ESP32 Module**
Please provide specifications of your module
- RAM/PSRAM: non
- Flash Size: 4MB

**Software (please complete the following information if applicable)**
 - IDE and Version: Arduino 1.8.8
 - OS: Windows 10/Android 9
 - Client used to access the server: Chrome(72.0.3626.121) and Firefox(65.0.1) on Android

dorarosi, 2019-03-21:
<pre>
<font color="#5e6d03">#include</font> <font color="#434f54">&lt;</font><b><font color="#d35400">WiFi</font></b><font color="#434f54">.</font><font color="#000000">h</font><font color="#434f54">&gt;</font>
<font color="#5e6d03">#include</font> <font color="#434f54">&lt;</font><font color="#000000">SPIFFS</font><font color="#434f54">.</font><font color="#000000">h</font><font color="#434f54">&gt;</font>
<font color="#5e6d03">#include</font> <font color="#434f54">&lt;</font><font color="#000000">FS</font><font color="#434f54">.</font><font color="#000000">h</font><font color="#434f54">&gt;</font>
<font color="#5e6d03">#include</font> <font color="#434f54">&lt;</font><b><font color="#d35400">HTTPSServer</font></b><font color="#434f54">.</font><font color="#000000">hpp</font><font color="#434f54">&gt;</font>
<font color="#5e6d03">#include</font> <font color="#434f54">&lt;</font><b><font color="#d35400">SSLCert</font></b><font color="#434f54">.</font><font color="#000000">hpp</font><font color="#434f54">&gt;</font>
<font color="#5e6d03">#include</font> <font color="#434f54">&lt;</font><b><font color="#d35400">HTTPRequest</font></b><font color="#434f54">.</font><font color="#000000">hpp</font><font color="#434f54">&gt;</font>
<font color="#5e6d03">#include</font> <font color="#434f54">&lt;</font><b><font color="#d35400">HTTPResponse</font></b><font color="#434f54">.</font><font color="#000000">hpp</font><font color="#434f54">&gt;</font>
<font color="#5e6d03">#include</font> <font color="#434f54">&lt;</font><font color="#000000">string</font><font color="#434f54">&gt;</font>

<font color="#00979c">const</font> <font color="#00979c">char</font><font color="#434f54">*</font> <font color="#000000">ssid</font> <font color="#434f54">=</font> <font color="#005c5f">&#34;OnePlus&#34;</font><font color="#000000">;</font>
<font color="#00979c">const</font> <font color="#00979c">char</font><font color="#434f54">*</font> <font color="#000000">password</font> <font color="#434f54">=</font> <font color="#005c5f">&#34;62415308!&#34;</font><font color="#000000">;</font>

<font color="#000000">httpsserver</font><font color="#434f54">:</font><font color="#434f54">:</font><b><font color="#d35400">HTTPSServer</font></b> <font color="#434f54">*</font> <font color="#000000">secureServer</font><font color="#000000">;</font>

<font color="#00979c">char</font> <font color="#000000">contentTypes</font><font color="#000000">[</font><font color="#000000">]</font><font color="#000000">[</font><font color="#000000">2</font><font color="#000000">]</font><font color="#000000">[</font><font color="#000000">32</font><font color="#000000">]</font> <font color="#434f54">=</font> <font color="#000000">{</font>
 &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;<font color="#000000">{</font><font color="#005c5f">&#34;.html&#34;</font><font color="#434f54">,</font> <font color="#005c5f">&#34;text&#47;html&#34;</font><font color="#000000">}</font><font color="#434f54">,</font>
 &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;<font color="#000000">{</font><font color="#005c5f">&#34;.css&#34;</font><font color="#434f54">,</font> &nbsp;<font color="#005c5f">&#34;text&#47;css&#34;</font><font color="#000000">}</font><font color="#434f54">,</font>
 &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;<font color="#000000">{</font><font color="#005c5f">&#34;.js&#34;</font><font color="#434f54">,</font> &nbsp;&nbsp;<font color="#005c5f">&#34;application&#47;javascript&#34;</font><font color="#000000">}</font><font color="#434f54">,</font>
 &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;<font color="#000000">{</font><font color="#005c5f">&#34;.json&#34;</font><font color="#434f54">,</font> <font color="#005c5f">&#34;application&#47;json&#34;</font><font color="#000000">}</font><font color="#434f54">,</font>
 &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;<font color="#000000">{</font><font color="#005c5f">&#34;.png&#34;</font><font color="#434f54">,</font> &nbsp;<font color="#005c5f">&#34;image&#47;png&#34;</font><font color="#000000">}</font><font color="#434f54">,</font>
 &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;<font color="#000000">{</font><font color="#005c5f">&#34;.jpg&#34;</font><font color="#434f54">,</font> &nbsp;<font color="#005c5f">&#34;image&#47;jpg&#34;</font><font color="#000000">}</font><font color="#434f54">,</font>
 &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;<font color="#000000">{</font><font color="#005c5f">&#34;.gif&#34;</font><font color="#434f54">,</font> &nbsp;<font color="#005c5f">&#34;image&#47;gif&#34;</font><font color="#000000">}</font><font color="#434f54">,</font>
 &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;<font color="#000000">{</font><font color="#005c5f">&#34;.wma&#34;</font><font color="#434f54">,</font> &nbsp;<font color="#005c5f">&#34;audio&#47;x-ms-wma&#34;</font><font color="#000000">}</font><font color="#434f54">,</font>
 &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;<font color="#000000">{</font><font color="#005c5f">&#34;&#34;</font><font color="#434f54">,</font> <font color="#005c5f">&#34;&#34;</font><font color="#000000">}</font>
<font color="#000000">}</font><font color="#000000">;</font>

<font color="#95a5a6">&#47;**</font>
<font color="#95a5a6"> * This function will either read the certificate and private key from SPIFFS or</font>
<font color="#95a5a6"> * create a self-signed certificate and write it to SPIFFS for next boot</font>
<font color="#95a5a6"> *&#47;</font>
[... cut ...]
 &nbsp;&nbsp;&nbsp;<font color="#000000">}</font>
<font color="#000000">}</font>

<font color="#00979c">void</font> <font color="#5e6d03">loop</font><font color="#000000">(</font><font color="#000000">)</font><font color="#000000">{</font>
 &nbsp;&nbsp;&nbsp;<font color="#000000">secureServer</font><font color="#434f54">-</font><font color="#434f54">&gt;</font><font color="#5e6d03">loop</font><font color="#000000">(</font><font color="#000000">)</font><font color="#000000">;</font>
<font color="#000000">}</font>

</pre>

fhessel, 2019-03-21:
> **Actual Behavior**
> I get the HTML code as plaintext startet with
> `Content-Type: text/html`
> `Connection: close`

Usually you see headers in the web view of the browser when the header _before_ those visible headers is corrupted, e.g. its value contains a line break.

Does this only happen for Android, or are you able to reproduce it in a desktop browser? If so, could you additionally provide the whole raw response from the developer tools, starting with `200 OK` or something similar?

dorarosi, 2019-03-22:
I will try to recreate the error on a desktop browser but my previouse attampts to use Chrome or Firefox on a desktop resulted in a crash of the esp32. I also tryed to use Safari on an iPhone and this displayed also nothing.

fhessel, 2019-03-23:
This seems to be a bug within the headers implementation. Running your code and accessing the server with curl returns something like this:
```
> GET /index.html HTTP/1.1
> Host: ...
> User-Agent: curl/7.55.1
> Accept: */*
> 
< HTTP/1.1 200 OK
< Content-Length: o open directory
< Content-Type: text/html
< 
```

So I'll need to fix the `HTTPHeaders` implementation in a way that actually copies the values until the response is sent, that's a bit less memory-saving but also a lot less error-prone. I'll rename the issue to reflect the actual problem. Might take some days, though. `HTTPHeaders` is used in various places and I need to verify that I don't cause memory leaks with the fix.

fhessel, 2019-03-26:
I checked again, the main problem wasn't the way the strings are organized, but I revised it anyway. The progress can be found in [this branch](https://github.com/fhessel/esp32_https_server/tree/issue37), if you want to check that out.

More importantly, I also found an error in the REST API example, which is also present in your sketch.

Could you try to change this part of your code …
```c++
        // Set length
        res->setHeader("Content-Length", "" + file.size());
```

… to this …

```c++
        // Set length
        res->setHeader("Content-Length", httpsserver::intToString(file.size()));
```

… and also add an …

```c++
#include <util.hpp>
```

… at the top of your sketch?

`"" + some int` will just not work, I might somehow got a bit confused with Java when writing that line in the example. Using the usual string stream isn't feasible on the ESP as it costs way to much program space for just converting an int to a string. At least I didn't get it running.

A short feedback would be appreciated, so I can close the issue if your problem is solved.

fhessel, 2019-03-30:
I merged the changes in the master branch, they are included in `v0.2.4`. If you have any further issues after changing the code snippet, just let me know!

closed 2019-03-30 by fhessel

dmarc1234, 2023-03-20:
Hi @fhessel 

I am still getting a a LoadProhibited exception when delivering larger pages as decribed by @dorarosi above.

I have checked the code and the above changes are already implemented in the library (v1.0.0), any ideas ?