fhessel/esp32_https_server #40: openssl/ssl.h: No such file or directory

ateeb327, 2019-06-25:
While uploadin code, it presents this error:
```
Arduino\libraries\esp32_https_server-master\src/HTTPResponse.hpp:12:25: fatal error: 
openssl/ssl.h: No such file or directory

Compilation Terminated
```

fhessel, 2019-06-25:
"openssl/ssl.h" is [part of the ESP-IDF](https://github.com/espressif/esp-idf/blob/master/components/openssl/include/openssl/ssl.h), on top of which the ESP32 Arduino Environment is built.

If the IDE says it's missing during compilation, that's most likely a missconfiguration of the ESP32 Arduino tools. Have you checked/tried reinstalling them? Can you build other sketches that – for example – use WiFiClientSecure? Which version of the IDE and of the Arduino Tools are you using?

fhessel, 2019-07-04:
Could you solve your problem or do you need further assistance to deal with this? Otherwise I'd close the issue.

ateeb327, 2019-07-08:
Actually I was trying this on ESP8266 :o:

fhessel, 2019-07-08:
Okay, that at least explains the error message.

Even though this library targets the Arduino Core for the ESP32, internally it uses some ESP-IDF modules directly to do communication and crypto, so this cannot work on the ESP8266, and I currently have no plans in porting it back to the ESP8266. The TLS performance of the 8266 is not that great anyway, and as the main goal for this lib was to support server-side TLS, porting it does not make much sense for me. For that reason I'd close this for now.

I'm sorry I cannot help you there.

closed 2019-07-08 by fhessel

Ulf1707, 2024-12-23:
I have the same message than ateeb327:

`In file included from Static_Page\Static_Page.ino:32:
Arduino\libraries\ESP32_HTTPS_Server\src/HTTPSServer.hpp:11:10: fatal error: openssl/ssl.h: No such file or directory`

May be I am wrong, but as far as I can see "openssl/ssl.h" is part of the ESP-IDF up to version 4.4. I am using version 5.4. 
My setup is esp32doit-devkit-v1 from esp32\hardware\esp32\3.0.7 with Arduino IDE 2.3.4 under Windows.
What can I do to get rid of this error?

Ulf1707, 2024-12-25:
I switched to Espressif's ESP-IDF V2.0.17 and changed in src/HTTPConnection.hpp

`#include <hwcrypto/sha.h>

with

#include <sha/sha_parallel_engine.h>`

now it compiles.

Battlecake91, 2025-01-11:
I have the same issue after trying to change my board to a ESP32-C3-SuperMini, now I cannot compile any project with HTTPS_Server in. 
I tried to take everything back, but even other projects with this lib doesn't work now. 

I don't know what the heck is going on and what configuration could be corrupted now.

But as Ulf1707 mentioned, the openssl/ssl.h isn't in the esp-idf above 4.4 - so this isn't a configuration error at least?

woidoi, 2025-02-20:
It's gone since version 5.0, according to espressif. 
I found it mentioned here- https://github.com/espressif/esp-idf/blob/release/v4.4/components/openssl/include/openssl/ssl.h
"#warning "OpenSSL component will be removed from ESP-IDF in v5.0, please use esp_tls instead"

Battlecake91, 2025-02-20:
Saw that too. I tried to change the include file, but it seems that this isn't easy to switch. Unfortunatelly.

woidoi, 2025-02-22:
I rolled back the ESP32 board selection from 3.1.1 to 1.0.6 and my project compiled and uploaded correctly. I was compiling for a Wemos Lolin32 Oled board that's been around for a while though.