fhessel/esp32_https_server #42: cert.h : no such file or directory

Patsjemoe, 2019-07-05:
Hi,
I donwloaded the esp32_https_server file, unzipped, ran script extras/create_cert.sh, but still get the error 
cert.h : no such file or directory, when compiling.
When looking in the directory, the cert.h is visibble. 
any suggestions ?
best regards
Ludo
[image]

fhessel, 2019-07-05:
I assume you are using the Arduino IDE (and not an alternative build environment like Platform IO)?

Then:
- Did you re-open the sketch after running the script?
- Does the IDE show cert.h and private_key.h as separate tabs? It should look like this when you've open the HTTPS and HTTP example:
[image]

Patsjemoe, 2019-07-05:
Hi Frank,
Yes, running Arduino (new to this, not very familiar yet. Used to program
in Bascom on AVR, Python on Raspberry Pi and Android.
I re-opened the sketch, even restarted arduino.
I *do not *see the cert.h and private key file ...
best regards
Ludo

fhessel, 2019-07-05:
Could you try adding them using _Sketch → Add File…_? As they're local project files and not part of the library, they need to be present in Arduino's tab bar, otherwise they're missing in your project and you'll get exactly the error you've mentioned.

create_cert.sh _should_ add the files to each example folder, and then the IDE _should_ pick them up when you open an example, but maybe something went wrong there.

Patsjemoe, 2019-07-05:
hi,
sketch -> add file , includes the file in your program, but not as tab....
I wonder if it is a setting in Arduino somewhere...
regards
Ludo

Patsjemoe, 2019-07-05:
when I start Arduino  from the directory (clicking on the https file) with
the cert.h file, I see the 2 files.
So it must be some setting of inclusion of files....
best regards

fhessel, 2019-07-05:
I don't know exactly how the IDE behaves if you open the files from the Explorer.

Opening the examples from the IDE (_File → Examples → ESP32 HTTP(S) Webserver → ..._, make sure you selected an ESP32 board before) should add all .ino, .h, .c, ... files from an example directory to a fresh project.

Patsjemoe, 2019-07-05:
as it works this way, I can find out later why Arduino is doing this....
Thanks for your help.
regards
Ludo

Patsjemoe, 2019-07-05:
Hi Frank,
sorry to be back.
When starting the program, I opened the serial monitor, and see it goes to
set up Wifi......................and keeps giving........................
what can I check to see why it does not connect ?
(i have added SSID and pSW)
best regards
Ludo

Patsjemoe, 2019-07-05:
Frank,
forget the question.
in the psw field, there were still <> left between " "
regards
Ludo

fhessel, 2019-07-06:
Any other questions/issures regarding setup? Otherwise I'd close this issue for now.

If some new problem should arise, just open another one and let me know.

Patsjemoe, 2019-07-06:
Do you close it, or do I have to do that ?
Thanks
Regards

fhessel, 2019-07-06:
Well, both of us can. I'd just do it then if you don't have any objections.

closed 2019-07-06 by fhessel