fhessel/esp32_https_server #43: Self-Signed-Certificate - Connect to WiFi Fails

flywire, 2019-07-08:
**Describe the bug**
Self-Signed-Certificate - Connect to WiFi Fails

**How To Reproduce**
Steps to reproduce the behavior:
1. Upload Self-Signed-Certificate
2. Setting up WiFi - never connects (see output below)

**Expected Behavior**
Expected `Connected. IP=`

**Actual Behavior**
Never connects:
```
Creating a new self-signed certificate.
This may take up to a minute, so be patient ;-)
Creating the certificate was successful
Setting up WiFi
.....
```
**ESP32 Module**
Please provide specifications of your module
```
BN: Unknown board
VID: 10C4
PID: EA60
```
```
C:\Users\user\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\2.6.1>esptool flash_id
esptool.py v2.6
Found 1 serial ports
Serial port COM3
Connecting........_____....._____....._____....._____.....__
Detecting chip type... ESP32
Chip is ESP32D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme
None
MAC: 00:00:00:00:00:00
Uploading stub...
Running stub...
Stub running...
Manufacturer: 20
Device: 4016
Detected flash size: 4MB
```
**Software (please complete the following information if applicable)**
 - IDE and Version: Arduino 1.8.9
 - OS: Windows 7 Home Premium
 - Client used to access the server: Arduino Serial console

flywire, 2019-07-08:
How should `WiFi.h` be configured?

flywire, 2019-07-08:
Not working again - tryed to run example version rather than one with configured WiFi. It would be useful to add some error checking to make sure wifi is working and configured.

fhessel, 2019-07-08:
I didn't want to over-complicate the examples by adding reconnect loops etc., they aren't meant for productive use but rather as executable documentation for the library itself. I assume that most users will have their own implementation for staying connected with the network, so there's no one-fits-all anyway.

If your WiFi is very unstable, try adding something like this to the main loop:
```c++
unsigned long last_reconnect = 0;
void loop() {
  if (WiFi.status() != WL_CONNECTED) {
    if (millis()-last_reconnect > 10000) {
      Serial.printf("Reconnecting to %s\n", SSID);
      WiFi.disconnect();
      delay(100);
      WiFi.begin(SSID, PSK);
      last_reconnect = millis();
    }
  }
}
```

If you did not specify any bind address when creating the server (optional parameters to the HTTPSServer() constructor), the server should survive that and accept requests on the WiFi interface once it is reconnected. If you used a specific bind address, you might need to `stop()`/`start()` the server again, but I'd consider that as an advanced use case.

> btw https://IP-address/ is giving me chrome error: NET::ERR_CERT_AUTHORITY_INVALID

That's expected behavior, as the examples only generate self-signed certificates. If you want valid certificates in every browser, you need a proper domain name for your ESP and a known public CA that signs your certificate for you. You can get that right with the lib, but you need to have an understanding of how certificates and certificate agencies work, and that's way beyond what I can explain here.

closed 2019-07-08 by flywire