fhessel/esp32_https_server #45: Error content-length header

r-zlotorzynski, 2019-07-13:
** My test environment  **
ESP32  WROVER 16

When trying to set a Content-length header such as 1, 10, 1000, 10000 ... the header is incorrectly formatted.

This function is ./src/util.cpp    (intToString)

```
std::string intToString(int i) {
  if (i==0) {
    return "0";
  }
  // We need this much digits
  int digits = ceil(log10(i));
  char c[digits+1];
  c[digits] = '\0';

  for(int x = digits-1; x >= 0; x--) {
    char v = (i%10);
    c[x] = '0' + v;
    i = (i-v)/10;
  }

  return std::string(c);
}
```

The results of this function are exemplary data:

[image]

For the above-mentioned numbers, the values are incorrect. In addition, no value is returned for length 1.

I have slightly changed the function on:
```
/**
    * TODO: Convert int to string C++
    *
    * @param            {int}                           i                              Int >= 0
    * @return             {std::string}
    * @public
*/
std::string intToString( int i ) {
    // Value is 0, return const 0
    if ( i <= 0 ) return std::string( "0" );

    // Prepare buffer
    char buffer[ 8 ];         

    // Copy INT to buffer
    itoa( i , buffer , 10 );

    // Return INT as C++ string
    return std::string( buffer );
}
```

After the change, the results look better

[image]

fhessel, 2019-07-13:
Thanks for finding this and the detailed information, it seems to be a typical off-by-one error.

The reason why I did not use `itoa` is the missing check for array bounds. Your example code will have problems with value bigger than 9.999.999 (even though that might be a theoretical problem only on the ESP32). You'd have to use the right length there too, which will make you do the same computation with `log10` that I did (or go for the maximum for type int, but I don't like to include architecture-dependent fixed numbers).

I'll have a look at it and fix the bug, but I don't have hardware for testing it at hand right now.

r-zlotorzynski, 2019-07-13:
Thank you for your response.
So far, I have used small portions of data, which is why I did not take into account this problem.
Thank you for your help.

commit 90c81776 (2019-09-17): Fix off-by-one error in number formatting. Resolves #45
closed 2019-09-18 by fhessel