fhessel/esp32_https_server #48: createSelfSignedCert running out of stack

ikostoski, 2019-08-20:
**Describe the bug**
The createSelfSignedCert function from SSLCert.cpp crashes many times before occasionally generating the certificate.

**How To Reproduce**
Steps to reproduce the behavior:
1. Compile and upload REST-API.ino.
2. Watch the 'Guru Meditation' errors...

**Expected Behavior**
Self-signed certificate generation that works reliably.

**Actual Behavior**
'loopTask' Stack is overrun most of the time for 1024bit key, and always for 2048bit key

**ESP32 Module**
Please provide specifications of your module
- RAM/PSRAM: 520Kb/None
- Flash Size: 4MB
- Other special properties: 

**Software (please complete the following information if applicable)**
 - IDE and Version:  Arduino 1.8.9
 - OS: N/A
 - Client used to access the server: N/A

**Additional context**
There appears to be many mbedtls structures allocated on the stack in 'cert_write' function, which may contribute to the problem, or simply mbedtls need space on stack to work with. 
Workaround for reliable self-signed certificate generation seems to be running it in separate task with huge stack (i.e. 16Kb), just for this purpose, and than deleting the task afterwards.

fhessel, 2019-08-20:
Thanks for the report and the analysis! When I implemented and tested it, I did experience the issue sometimes, but it was not as severe as you describe it.

I'll have a look at it. Maybe the time has come to make a background worker task for the server mandatory, as this would help to solve many other problems as well (like blocking handler functions etc.). Adjusting the stack size of the main task does not seem a good solution to me and is hard to do with the Arduino environment, anyway.

rljonesau, 2020-04-15:
Likewise I'm experiencing stack overflow generating both 2048 bit & 1024bit keys.

1024bit key crashes much more quickly than 2048, which suggests an issue towards the end of the process?

mc-hamster, 2020-10-07:
I have just run into this bug. It happens every time for me in my application. Happy to share information if anyone would like.

mc-hamster, 2020-10-08:
We're now using this library in Meshtastic ( https://github.com/meshtastic/Meshtastic-device ) and the bug described in this issue is 100% reproducible there. I used the work around by the original poster and that got me unblocked.

FStefanni, 2021-06-15:
Hi,

I can confirm the same issue of stack overflow..
I use it in a more complex setup w.r.t. the rest api example sketch, but nevertheless it should work...

In my case, the overflow happens even with 1024 bits cert.

I had no time to check the code, so I have no idea if it is due to a recursive function call, or to a lot of allocations on the stack (or few huge stack allocations or else).

Regards.