fhessel/esp32_https_server #49: Is there a fancy way to modify constants without touch the libary files?

ricard2k, 2019-08-21:
**Describe Your Goal**
I need to modify one constant in HttpServerConstants  but I dont want to edit the library file, I want to do it in my code.

**What Does Your Project Look Like**
I need to increase the max line size because I'm using JWT tokens (large string in Authorization header) as authentication.

If you can boil your question down to a [minimal working example]
I need to get with a large header as in this sample:
 
``curl -X GET \
  https://192.168.6.73/relay1/0 \
  -H 'Authorization: Bearer aaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaa.bbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbb-ccccccccccccccccccccccccccccccccccccccc-dddddddddddddddddddddddddddddddddddddddddddddddddd-eeeeeeeeeeeeeeeeeeeeeee-fffffffffffffffffffffffffffff-ggggggggggggggg-hhhhhhhhhhhhhhhhhhhhhhhhh..............................'
```
and at the MCU I get the next output when the request is being parsed:

```
1092994 HTTPSServer->debug: [-->] New connection. Socket fid is:  0x38
1095584 HTTPSServer->debug: [   ] There is data on the connection socket. fid= 0x38
1095586 HTTPSServer->debug: [   ] Request line finished: method=GET, resource=/relay1/0
1095590 HTTPSServer->debug: [   ] There is data on the connection socket. fid= 0x38
1095595 HTTPSServer->debug: [ERR] Line length exceeded. Server error.
Server error
1095604 HTTPSServer->debug: [<--] Connection has been closed. fid =  0x38
1095610 HTTPSServer->debug: [   ] Free headers
```

**ESP32 Module** 
ESP WROOM32
- Flash Size: 8M

**Software (please complete the following information if applicable)**
 - IDE and Version: PlatformIO 4.0.0
 - Client used to access the server: simple cURL

fhessel, 2019-08-21:
I'm working on that, I already allowed modifying the log level and added that now also for the other constants, but only on the develop-branch (see the referenced commit).

In your platform.ini, you then can use something like this:

```ini
[env:esp32dev]
platform = espressif32
board = esp32dev
framework = arduino
build_flags =
  -DHTTPS_REQUEST_MAX_HEADER_LENGTH=768
```

If the develop branch isn't stable enough, you can temporarily make the #ifndef/#define/#endif changes in your local copy of the repo, as I'll merge that soon.

Does that solve your problem?

ricard2k, 2019-08-21:
Thanks a lot! This what I was looking for!

closed 2019-08-21 by ricard2k