fhessel/esp32_https_server #50: HTTPServer crashes where stopping and starting the server

pacoandres, 2019-09-16:
I think it's a pull request but github does not allow me to create it.

I'm working on a project that needs to stop the server for doing a long task and then restart it again.
When I do this and call server.loop after restarting, the program crashes with a guru meditation.

I've seen that it happens when the code tries to delete an already deleted pointer, so it can be solved by assigning NULL to all the deleted pointers in the stop function.

Here is the diff output for changing it in a new release:

`*** HTTPServer.cpp.old  2019-09-16 15:26:25.000000000 +0200
--- HTTPServer.cpp      2019-09-16 15:27:56.000000000 +0200
***************
*** 69,74 ****
--- 69,75 ----
            // and wait for the client
            if (_connections[i]->isClosed()) {
              delete _connections[i];
+             _connections[i] = NULL;
            } else {
              hasOpenConnections = true;
            }
***************
*** 208,211 ****
  }
  
  } /* namespace httpsserver */
-  
--- 209,211 ----`
Everything else works fine. Tank you very much.

fhessel, 2019-09-17:
I couldn't test it yet, but it looks like a valid suggestion. I'll test it later.

If you want to create a real pull request from it (so you're mentioned as the contributor), you could do so by:
1. Forking the repository (button is on the top-right of this page)
2. Add the change to your own repository
3. Come back to this repo and file the PR

If not, I also can add this change. Let me know what you prefer

pacoandres, 2019-09-17:
I think it could be easier if you do the change.

And thanks for the library again. I like it very much.

fhessel, 2019-09-17:
I've tested it and could verify the problem, and also your solution. Thank you for pointing that out!

It's commited to the develop branch. I'll try to bundle it with some other fixed and create a release soon. Commit 1ca85e9 should be building and working, if you want it faster.

commit 1ca85e91 (2019-09-17): Clean close connections correctly when stopping the server. Closes #50
closed 2019-09-18 by fhessel