fhessel/esp32_https_server #53: Exception when decoding URL parameters

vincent-bruel, 2019-10-23:
First: thanks for your great work.
Problem is: ESP32 restart with Backtrace in console when pushing a button with a formaction URL like this one:
GET /coucou/Supprimer/2? 

Viewing the backtrace (decoded, see below) It seems that there is a problem trying to get a parameter that doesn't exist (after the character '?'), so it should be a simple out of index problem, but I may be wrong.

Steps to reproduce:
1. Serve the following file via a handleRoot
<!DOCTYPE html>
<html>
<body>

<h2>Basic HTML Table</h2>
<form>
<table style="width:100%">
  <tr>
    <th>Firstname</th>
    <th>Lastname</th> 
    <th>Age</th>
  </tr>
  <tr>
    <td><input type="submit" value="XXX"  formaction="coucou/Supprimer/2"></td>
    <td>Smith</td>
    <td>50</td>
  </tr>
  <tr>
    <td>Eve</td>
    <td>Jackson</td>
    <td>94</td>
  </tr>
  <tr>
    <td>John</td>
    <td>Doe</td>
    <td>80</td>
  </tr>
</table>
</form>
</body>
</html>

3. On the client, click on the XXX button

4. See error in console:
23:54:28.701 -> [HTTPS:I] New connection. Socket FID=59
23:54:28.701 -> [HTTPS:I] Request: GET /posologie/Supprimer/2? (FID=58)
23:54:28.736 -> abort() was called at PC 0x401811bf on core 1
23:54:28.736 -> 
23:54:28.736 -> Backtrace: 0x4008c454:0x3ffb1740 0x4008c685:0x3ffb1760 0x401811bf:0x3ffb1780 0x40181206:0x3ffb17a0 0x4016ebd3:0x3ffb17c0 0x401705ef:0x3ffb17e0 0x400dd82e:0x3ffb1a90 0x400dd9c3:0x3ffb1ab0 0x400ddf16:0x3ffb1ae0 0x400dbe05:0x3ffb1c30 0x400dd532:0x3ffb1df0 0x400d76d5:0x3ffb1e20 0x400d707e:0x3ffb1e40 0x400e4909:0x3ffb1fb0 0x40088b9d:0x3ffb1fd0
23:54:28.770 -> 
23:54:28.770 -> Rebooting...

Backtrace decoded below:
Decoding stack results
0x4008c454: invoke_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c line 155
0x4008c685: abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c line 170
0x401811bf: __cxxabiv1::__terminate(void (*)()) at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/eh_terminate.cc line 47
0x40181206: std::terminate() at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/eh_terminate.cc line 57
0x4016ebd3: __cxxabiv1::__cxa_throw(void*, std::type_info*, void (*)(void*)) at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/eh_throw.cc line 87
0x401705ef: std::__throw_out_of_range_fmt(char const*, ...) at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/src/c++11/functexcept.cc line 104
0x400dd82e: std::vector   , std::allocator  >, std::allocator   , std::allocator  > > >::_M_range_check(unsigned int) const at c:\users\vb\appdata\local\arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0\xtensa-esp32-elf\include\c++\5.2.0\bits/stl_vector.h line 803
0x400dd9c3: httpsserver::ResourceParameters::setUrlParameter(unsigned char, std::__cxx11::basic_string  , std::allocator  > const&) at c:\users\vb\appdata\local\arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0\xtensa-esp32-elf\include\c++\5.2.0\bits/stl_vector.h line 824
0x400ddf16: httpsserver::ResourceResolver::resolveNode(std::__cxx11::basic_string  , std::allocator  > const&, std::__cxx11::basic_string  , std::allocator  > const&, httpsserver::ResolvedResource&, httpsserver::HTTPNodeType) at C:\Users\vb\Documents\Arduino\libraries\libraries\esp32_https_server-master\src\ResourceResolver.cpp line 126
0x400dbe05: httpsserver::HTTPConnection::loop() at C:\Users\vb\Documents\Arduino\libraries\libraries\esp32_https_server-master\src\HTTPConnection.cpp line 426
0x400dd532: httpsserver::HTTPServer::loop() at C:\Users\vb\Documents\Arduino\libraries\libraries\esp32_https_server-master\src\HTTPServer.cpp line 122
0x400d76d5: ServerWeb::loop() at C:\Users\vb\AppData\Local\Temp\arduino_build_9243\sketch\ServerWeb.cpp line 36

**ESP32 Module**
Please provide specifications of your module
Regular dev module like LOLIN32
- RAM/PSRAM: no PSRAM
- Flash Size: 4Mo
- Other special properties: 

**Software (please complete the following information if applicable)**
 - IDE and Version: tested both on VSCODE+Platformio and ArduinoIDE 1.8.10
 - OS: Windows 10 (updated)
 - Client used to access the server: google chrome

**Additional context**
EspExceptionDecoder to decode the stacktrace.

fhessel, 2019-10-24:
That looks indeed like an index problem. In [ResourceResolver.cpp](https://github.com/fhessel/esp32_https_server/blob/master/src/ResourceResolver.cpp#L43), it is assumed that there are actually parameters when the `?` is present. Thanks for the report!

I'll see what I can do, but it may take until the latest PR is merged until the fix gets to the master branch.

fhessel, 2019-10-24:
Do you have an additional wildcard route defined? Like `/coucou/Supprimer/*` or something similar? If I take the _Static Page_ example from the library and just replace `handleRoot(...)` like shown as follows, I cannot reproduce the problem:

```c++
void handleRoot(HTTPRequest * req, HTTPResponse * res) {
  // Status code is 200 OK by default.
  // We want to deliver a simple HTML page, so we send a corresponding content type:
  res->setHeader("Content-Type", "text/html");
  // The response implements the Print interface, so you can use it just like
  // you would write to Serial etc.
  res->print("<!DOCTYPE html>\n");
  res->print("<html>\n");
  res->print("<body>\n");
  res->print("\n");
  res->print("<h2>Basic HTML Table</h2>\n");
  res->print("<form>\n");
  res->print("<table style=\"width:100%\">\n");
  res->print("  <tr>\n");
  res->print("    <th>Firstname</th>\n");
  res->print("    <th>Lastname</th> \n");
  res->print("    <th>Age</th>\n");
  res->print("  </tr>\n");
  res->print("  <tr>\n");
  res->print("    <td><input type=\"submit\" value=\"XXX\"  formaction=\"coucou/Supprimer/2\"></td>\n");
  res->print("    <td>Smith</td>\n");
  res->print("    <td>50</td>\n");
  res->print("  </tr>\n");
  res->print("  <tr>\n");
  res->print("    <td>Eve</td>\n");
  res->print("    <td>Jackson</td>\n");
  res->print("    <td>94</td>\n");
  res->print("  </tr>\n");
  res->print("  <tr>\n");
  res->print("    <td>John</td>\n");
  res->print("    <td>Doe</td>\n");
  res->print("    <td>80</td>\n");
  res->print("  </tr>\n");
  res->print("</table>\n");
  res->print("</form>\n");
  res->print("</body>\n");
  res->print("</html>\n");
}
```

However, I only have Chrome and Firefox on Linux for testing, so I cannot preclude that the cause is somehow related to that. Interestingly, only Chrome creates that additional question mark, Firefox doesn't.

vincent-bruel, 2019-10-25:
Hi,
Yes, sorry I forgot to mention that:
        ResourceNode *nodeSupprimerLigne = new ResourceNode("/*/Supprimer/*", "GET", &ServerWeb::handleSupprimerLigne);

vincent-bruel, 2019-10-25:
Previous message is wrong, cause I've forgot to put the code into backslash so the first star disapeared.
       URL pattern is: slash star slash Supprimer slash star
`ResourceNode *nodeSupprimerLigne = new ResourceNode("/*/Supprimer/*", "GET", &ServerWeb::handleSupprimerLigne);`

vincent-bruel, 2019-10-25:
My project is too big to cut it into pieces.
Anyway, to reproduce the bug in a very similar environment, I've started from the Self signed certificate example, and added the following code line 121 just before secureServer->start() in order to create my own Wifi Access point.
"Root" node generates an html page but the X button generating the URL.
"S" node generates the same html code with another title.
And I cannot reproduce the bug neither with the following code:

`/**
 * Example for the ESP32 HTTP(S) Webserver
 *
 * IMPORTANT NOTE:
 * To run this script, you need to
 *  1) Enter your WiFi SSID and PSK below this comment
 *
 * This script will install an HTTPS Server on your ESP32 with the following
 * functionalities:
 *  - Show simple page on web server root
 *  - 404 for everything else
 * 
 * In contrast to the other examples, the certificate and the private key will be
 * generated on the ESP32, so you do not need to provide them here.
 * (this means no need to run create_cert.sh)
 */

// TODO: Configure your WiFi here
#define WIFI_SSID "your ssid"
#define WIFI_PSK  "your pwd"

// We will use wifi
#include <WiFi.h>

// Includes for the server
#include <HTTPSServer.hpp>
#include <SSLCert.hpp>
#include <HTTPRequest.hpp>
#include <HTTPResponse.hpp>

// The HTTPS Server comes in a separate namespace. For easier use, include it here.
using namespace httpsserver;

SSLCert * cert;
HTTPSServer * secureServer;

// Declare some handler functions for the various URLs on the server
void handleS(HTTPRequest * req, HTTPResponse * res);
void handleRoot(HTTPRequest * req, HTTPResponse * res);
void handle404(HTTPRequest * req, HTTPResponse * res);

void setup() {
  // For logging
  Serial.begin(115200);
  delay(3000); // wait for the monitor to reconnect after uploading.

  Serial.println("Creating a new self-signed certificate.");
  Serial.println("This may take up to a minute, so be patient ;-)");

  // First, we create an empty certificate:
  cert = new SSLCert();

  // Now, we use the function createSelfSignedCert to create private key and certificate.
  // The function takes the following paramters:
  // - Key size: 1024 or 2048 bit should be fine here, 4096 on the ESP might be "paranoid mode"
  //   (in generel: shorter key = faster but less secure)
  // - Distinguished name: The name of the host as used in certificates.
  //   If you want to run your own DNS, the part after CN (Common Name) should match the DNS
  //   entry pointing to your ESP32. You can try to insert an IP there, but that's not really good style.
  // - Dates for certificate validity (optional, default is 2019-2029, both included)
  //   Format is YYYYMMDDhhmmss
  int createCertResult = createSelfSignedCert(
    *cert,
    KEYSIZE_2048,
[... 150 lines cut ...]
  res->print("</table>\n");
  res->print("</form>\n");
  res->print("</body>\n");
  res->print("</html>\n");
}

void handle404(HTTPRequest * req, HTTPResponse * res) {
  // Discard request body, if we received any
  // We do this, as this is the default node and may also server POST/PUT requests
  req->discardRequestBody();

  // Set the response status
  res->setStatusCode(404);
  res->setStatusText("Not Found");

  // Set content type of the response
  res->setHeader("Content-Type", "text/html");

  // Write a tiny HTTP page
  res->println("<!DOCTYPE html>");
  res->println("<html>");
  res->println("<head><title>Not Found</title></head>");
  res->println("<body><h1>404 Not Found</h1><p>The requested resource was not found on this server.</p></body>");
  res->println("</html>");
}`

vincent-bruel, 2019-10-25:
Hi,
I've made a patch to HTTPConnection.cpp, adding this code after line 374 to remove the trailing '?'
       ` if (_httpResource.find('?') == _httpResource.size() - 1) {
            Serial.println("Removing trailing '?' at the end of the request");
            _httpResource = _httpResource.substr(0, _httpResource.size() - 1);
        }`

After that my URL = /contacts/Supprimer/3
should match a ResourceNode defined with "/contacts/Supprimer/*"
But still an exception raised at line 118 in ResourceResolver.cpp, just after seeing
"Testing parameter match on /contacts/Supprimer/*" in the console:

Decoding stack results
0x4008c454: invoke_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c line 155
0x4008c685: abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c line 170
0x40181b33: __cxxabiv1::__terminate(void (*)()) at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/eh_terminate.cc line 47
0x40181b7a: std::terminate() at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/eh_terminate.cc line 57
0x4016f547: __cxxabiv1::__cxa_throw(void*, std::type_info*, void (*)(void*)) at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/eh_throw.cc line 87
0x40170f63: std::__throw_out_of_range_fmt(char const*, ...) at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/src/c++11/functexcept.cc line 104
0x400ddd72: std::vector   , std::allocator  >, std::allocator   , std::allocator  > > >::_M_range_check(unsigned int) const at c:\users\vb\appdata\local\arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0\xtensa-esp32-elf\include\c++\5.2.0\bits/stl_vector.h line 803
0x400ddf07: httpsserver::ResourceParameters::setUrlParameter(unsigned char, std::__cxx11::basic_string  , std::allocator  > const&) at c:\users\vb\appdata\local\arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0\xtensa-esp32-elf\include\c++\5.2.0\bits/stl_vector.h line 824
0x400de46a: httpsserver::ResourceResolver::resolveNode(std::__cxx11::basic_string  , std::allocator  > const&, std::__cxx11::basic_string  , std::allocator  > const&, httpsserver::ResolvedResource&, httpsserver::HTTPNodeType) at C:\Users\vb\Documents\Arduino\libraries\libraries\esp32_https_server-master\src\ResourceResolver.cpp line 118
0x400dc212: httpsserver::HTTPConnection::loop() at C:\Users\vb\Documents\Arduino\libraries\libraries\esp32_https_server-master\src\HTTPConnection.cpp line 435
0x400dda76: httpsserver::HTTPServer::loop() at C:\Users\vb\Documents\Arduino\libraries\libraries\esp32_https_server-master\src\HTTPServer.cpp line 122
0x400d7711: ServerWeb::loop() at C:\Users\vb\AppData\Local\Temp\arduino_build_346644\sketch\ServerWeb.cpp line 36

vincent-bruel, 2019-10-29:
The bug line 118 of ResourceResolver.cpp is also activated if I define the resource node with a star:
`ResourceNode *nodeSupprimerLigneContacts = new ResourceNode("/contacts/Supprimer/*", "GET", &ServerWeb::handleSupprimerLigne);
`
And activate it with an URL like /contacts/Supprimer/3

fhessel, 2019-11-03:
I did a complete rewrite of the resource resolving. Could you give a try to this branch and let me know if that solves your problem?
https://github.com/fhessel/esp32_https_server/tree/issue-53

I think it had to do with the usage of `capacity` and `size` in `ResourceParameters::setUrlParameter`. But I rewrote the whole thing, because I did only consider URL parameters at the end (so `/route/*/*` but not `/*/action/*`) before, and that's no longer the case. You can now put the placeholder anywhere you like, as long as it's enclosed by slashes or at the end of the pattern. So `/*/Supprimer/*` should be perfectly fine.

vincent-bruel, 2019-11-03:
Hi, that's great news :) !
I will give it a try ASAP and let you know the result here.
Thanks very much anyway for your help and your library.

vincent-bruel, 2019-11-04:
It seems to work very well.
I've tested the following URLs in a ResourceNode with success:
`"/*/xxx"
"/*/yyy/*"
`
Thanks

fhessel, 2019-11-04:
Thanks for the quick testing. I'll queue the change as a PR and include it in the next release. You should be able to work with the branched version of the lib until then, as the fix is the only difference to `master`/`develop`.

vincent-bruel, 2019-11-04:
OK, that's exactly what I've planned to do.

PR #54 (2019-12-24): Rewrite and Simplify Resource Resolving
The resource resolver was quite complex and had issues when dealing with URL parameters containing placeholders not only at the end. Also, the combination of URL parameters (like `/my/endpoint/*`) in combination with request parameters (`/my/endpoint/42?some=value`) wasn't handled well.

This PR contains a rewrite of the resource resolver that tackles these issues.

**References**

- Resolves #53
closed 2019-12-25 by fhessel