fhessel/esp32_https_server #55: Store Certificate and Private Key in Preferences

icubex, 2019-11-04:
Can the Certificate and Private Key be stored in Preferences ? We tried the code shown partially/excerpted below but the ESP32 crashes after connecting to WiFi and starting the server successfully, with a message that "setupCert failed". We can't find this function anywhere. The cert/key seem to be retrieved correctly from memory as they are identically printed via Serial. If the cert/key are not retrieved from Preferences, ie. freshly created, there is no crash. Why does it crash using the retrieved cert/key and not with the freshly created one ?

```c++
#include <Preferences.h>
#include <WiFi.h>
#include <WiFiManager.h>
#include <sstream>
#include <HTTPSServer.hpp>
#include <SSLCert.hpp>
#include <HTTPRequest.hpp>
#include <HTTPResponse.hpp>
#include <WebsocketHandler.hpp>

#if CONFIG_FREERTOS_UNICORE
#define ARDUINO_RUNNING_CORE 0
#else
#define ARDUINO_RUNNING_CORE 1
#endif

Preferences prefs;
WiFiManager wifiManager;
using namespace httpsserver;
SSLCert * cert;
HTTPSServer * secureServer;

void serverTask(void *params) {

  ResourceNode * nodeRoot    = new ResourceNode("/", "GET", &handleRoot);
  ResourceNode * node404     = new ResourceNode("", "GET", &handle404);

  secureServer->registerNode(nodeRoot);
  secureServer->setDefaultNode(node404);

  Serial.println("Starting server...");
  secureServer->start();
  if (secureServer->isRunning()) {
    Serial.print("Server ready. ");
    Serial.println(WiFi.localIP());

    while(true) {
      secureServer->loop();
      delay(1);
    }
  }
}

void setup() {

  size_t pkLen = prefs.getBytesLength("PK");
  size_t certLen = prefs.getBytesLength("cert");

  if (pkLen && certLen) {

    uint8_t pkBuffer[pkLen]; 
    prefs.getBytes("PK", pkBuffer, pkLen);
    
    uint8_t certBuffer[certLen];
    prefs.getBytes("cert", certBuffer, certLen);

    cert = new SSLCert(certBuffer, certLen, pkBuffer, pkLen);
    
    Serial.println("Retrieved Private Key " + String(cert->getPKLength()));
    for (int i = 0; i < cert->getPKLength(); i++)
      Serial.print(cert->getPKData()[i], HEX);
    Serial.println();

    Serial.println("Retrieved Certificate " + String(cert->getCertLength()));      
    for (int i = 0; i < cert->getCertLength(); i++)
      Serial.print(cert->getCertData()[i], HEX);
    Serial.println();
  
  } else {
  
    cert = new SSLCert();
     int createCertResult = createSelfSignedCert(
      *cert,
      KEYSIZE_2048,
      "CN=myesp32.local,O=FancyCompany,C=DE",
      "20190101000000",
      "20300101000000"
    );
  
    if (createCertResult != 0) {  

      Serial.printf("Creating the certificate failed. Error Code = 0x%02X, check SSLCert.hpp for details", createCertResult);
      while(true) delay(500);    

    } else {    

      Serial.println("Creating the certificate was successful");
      Serial.println("Private Key " + String(cert->getPKLength()));
      for (int i = 0; i < cert->getPKLength(); i++)
        Serial.print(cert->getPKData()[i], HEX);
      Serial.println();

      Serial.println("Certificate " + String(cert->getCertLength()));      
      for (int i = 0; i < cert->getCertLength(); i++)
        Serial.print(cert->getCertData()[i], HEX);
      Serial.println();    

      prefs.putBytes("PK", (uint8_t *)cert->getPKData(), cert->getPKLength());
      prefs.putBytes("cert", (uint8_t *)cert->getCertData(), cert->getCertLength());  
    }
  }

  secureServer = new HTTPSServer(cert);

  WiFi.mode(WIFI_STA); 
  wifiManager.setDebugOutput(true);
  wifiManager.autoConnect(MyESP32);
  
  xTaskCreatePinnedToCore(serverTask, "https443", 6144, NULL, 1, NULL, ARDUINO_RUNNING_CORE);
}

```

**ESP32 Module**
WEMOS LOLIN32
- RAM/PSRAM: 520 KiB SRAM
- Flash Size: 4MB

**Software**
 - IDE and Version: Arduino 1.8.10
 - OS: MacOS 10.14.6

**Crash log**
*WM: [1] Free heap:        118024
*WM: [1] ESP-IDF version:  v3.2.3-14-gd3e562907
*WM: [1] AutoConnect 
*WM: [2] ESP32 event handler enabled 
*WM: [2] Connecting as wifi client... 
*WM: [1] STA static IP:
*WM: [2] setSTAConfig static ip not set 
*WM: [3] WIFI station disconnect 
*WM: [1] Connecting to SAVED AP: MyISPXXX
*WM: [3] Using Password: ABCDEFGHIJKL
*WM: [3] WiFi station enable 
*WM: [1] connectTimeout not set, ESP waitForConnectResult... 
*WM: [2] Connection result: WL_CONNECTED
*WM: [3] lastconxresult: WL_CONNECTED
*WM: [1] AutoConnect: SUCCESS 
*WM: [1] STA IP Address: 192.168.2.23

MyESP32 connected via WiFi to SSID: MyISPXXX with IP: 192.168.2.23
Starting server...
setupCert failed
E (9122) FreeRTOS: FreeRTOS Task "https443" should not return, Aborting now!
abort() was called at PC 0x40088c7b on core 1

Backtrace: 0x4008c514:0x3ffe5b30 0x4008c745:0x3ffe5b50 0x40088c7b:0x3ffe5b70

fhessel, 2019-11-04:
I assume that the memory will be overridden, as you allocate it on the stack in the `setup()` function. The `SSLCert` constructor does not copy the memory, but relies on it being valid for the time the server is operating.

Could you try and move them to the heap?

```c++
// ...
  size_t pkLen = prefs.getBytesLength("PK");
  size_t certLen = prefs.getBytesLength("cert");

  if (pkLen && certLen) {

    uint8_t *pkBuffer = new uint8_t[pkLen];
    prefs.getBytes("PK", pkBuffer, pkLen);
    
    uint8_t *certBuffer = new uint8_t[certLen];
    prefs.getBytes("cert", certBuffer, certLen);

    cert = new SSLCert(certBuffer, certLen, pkBuffer, pkLen);
// ...
```

icubex, 2019-11-04:
:-) That's it, thanks !