fhessel/esp32_https_server #56: Wont work whne bluetooth is started

jmiranda7777, 2019-11-12:
Everything works great until I start

  pServer = BLEDevice::createServer();

I can include the files
#include <BLEDevice.h>
#include <BLEUtils.h>
#include <BLEServer.h>
and it works unless I start bluetooth.
There is some kind of conflict but i haven't been able to figure it out.
I am using partition min sips with OTA and I am only 88%. Its not crashes I see the message about web hits but there are errors starting ssl

fhessel, 2019-11-12:
Hi,

could you please provide some infos about your environment (Arduino
IDE/Platform IO) and the versions of the tools you're using?

And if you're using PlatformIO, could you please enable full logging on
DEBUG (see bottom of the repo's readme file) and attach the output?

jmiranda7777, 2019-11-12:
arduino ide 1.8.9
board nano32s
partition min spiff
I haven't done much 
Here is the Arduino code see if it reproduces the issue for you

[min32SSL.zip](https://github.com/fhessel/esp32_https_server/files/3837984/min32SSL.zip)

fhessel, 2019-11-12:
It's a memory issue. Every TLS connection needs roughly 40-50k of large blocks on the heap, which isn't free after initializing Bluetooth. I tested a minimal example on the WROOM-32 (the common DevBoard) and on a custom board with a [WROVER chip](https://eu.mouser.com/ProductDetail/espressif/esp32-wrover-b-16mb/?qs=w%2Fv1CP2dgqrsSzc2ZlAxuA==). You'll find the sketch and the output below. I think, the results will be similar for your project, even though I used PIO.

**Project Files**

<details>
<summary>main.cpp</summary>

```c++
#include <Arduino.h>

#define WIFI_SSID "..."
#define WIFI_PSK "..."

#include "cert.h"
#include "private_key.h"
#include "favicon.h"

#include <WiFi.h>

#include <HTTPSServer.hpp>
#include <SSLCert.hpp>
#include <HTTPRequest.hpp>
#include <HTTPResponse.hpp>

#include <BLEDevice.h>
#include <BLEUtils.h>
#include <BLEServer.h>

using namespace httpsserver;

SSLCert cert = SSLCert(
  example_crt_DER, example_crt_DER_len,
  example_key_DER, example_key_DER_len
);

HTTPSServer secureServer = HTTPSServer(&cert);

void handleRoot(HTTPRequest * req, HTTPResponse * res);
void handleFavicon(HTTPRequest * req, HTTPResponse * res);
void handle404(HTTPRequest * req, HTTPResponse * res);

unsigned long lastReconnect = 0;

void setup() {
  // For logging
  Serial.begin(115200);

  // Connect to WiFi
  Serial.println("Setting up WiFi");
  WiFi.begin(WIFI_SSID, WIFI_PSK);
  while (WiFi.status() != WL_CONNECTED) {
    if (millis()-lastReconnect > 10000) {
      lastReconnect = millis();
      WiFi.disconnect();
      delay(500);
      Serial.print(":");
      WiFi.begin(WIFI_SSID, WIFI_PSK);
    } else {
      delay(500);
      Serial.print(".");
    }
  }
  Serial.print("Connected. IP=");
  Serial.println(WiFi.localIP());

  ResourceNode * nodeRoot    = new ResourceNode("/", "GET", &handleRoot);
  ResourceNode * nodeFavicon = new ResourceNode("/favicon.ico", "GET", &handleFavicon);
  ResourceNode * node404     = new ResourceNode("", "GET", &handle404);
  secureServer.registerNode(nodeRoot);
  secureServer.registerNode(nodeFavicon);
[... 121 lines cut ...]
[HTTPS:I:     11296] Connection closed. Socket FID=55
Free Heap: 4338552 bytes
Starting BT...
Started BT...
Free Heap: 4246292 bytes
[HTTPS:I:     17702] New connection. Socket FID=56
Free Heap: 4204760 bytes
[HTTPS:I:     19971] Request: GET / (FID=56)
[HTTPS:I:     19984] Connection closed. Socket FID=56
Free Heap: 4245748 bytes
```

</details>

As it's clearly visible, on the WROOM, only 51356 bytes of heap memory remain after initializing BLE. That's not sufficient to do the TLS handshake, and there is nothing I can do about it, as it's an atomic call to the openSSL implementation of the underlying esp-idf.

You also can see that enabling PSRAM solves that problem on the WROVER.

What you can do:

- Use `HTTPServer` instead of `HTTPSServer`, but you don't have encryption and authentication then.
- Try smaller certificates (maybe I also give that a try later on), but you most likely won't get over a single "concurrent" connection.
- Use another ESP32 module. If you're designing custom hardware, you usually replace your DevBoard with a custom PCB at some time, so you can just pick an ESP module with additional RAM (and maybe also flash, to circumvent the problem with the huge BLE library). Meanwhile, you'll also find DevBoards with more memory, look for "PSRAM".

I know that answer's not very satisfying, so I you have another idea to discuss here, let me know. Otherwise I'd close the issue, as I cannot act on it meaningfully.

jmiranda7777, 2019-11-13:
You can close the bug.
Thank you so much for the answer I will look for a Dev Board with more working memory.
The SSL works really well. The bluetooth on the chip needs a lot of work its way too heavy.
Thank you for all your support.
Jorge

closed 2019-11-13 by fhessel