fhessel/esp32_https_server #63: Support full standard of query ResourceParameters

btittelbach, 2019-12-02:
**Is your feature request related to a problem? Please describe.**
There is currently now way to parse values for multiple parameters with the same key-name.
e.g. a valid query string could be: `?ctrarray=1&ctrarray=2&ctrarray=3&ctrarray=4` ... and `getRequestParameter` would always only return the first occurrence of the parameter.

**Describe the solution you'd like**
This is sometimes used in the wild and can be useful in bare-bone microcontroller webpage environments. Most frameworks support it.

**Describe alternatives you've considered**
Not supporting this, since most people hopefully won't need to parse queries like this.

fhessel, 2019-12-03:
Do you know anything more current/more official on this topic than what's discussed in [this answer on StackOverflow](https://stackoverflow.com/questions/1746507/authoritative-position-of-duplicate-http-get-query-keys)?

Returning a collection of strings from `getRequestParameter` would make processing the request parameters more complicated in most cases (at least that's what I'd assume), where this is not needed, and if the specification does not cover the way this should be handled, I'd not make the API more complicated than that.

Maybe it would be a solution to include the duplicate parameters when using the iterator from #62 to go through the request parameters? In that case the user can decide if duplicate values are relevant and define how they should be handled.

btittelbach, 2019-12-03:
Nothing more authoritative that that. As it says, there does not seem to be a standard, thus everything that the schema allows is used somewhere, including the use of multiple identical keys. I know that a decade ago already some frameworks would translate array data to request-data in this form.

I think that mostly this will not be needed, so just including them in the iterator from #62 should be perfectly fine.

commit 2f50357b (2019-12-05): Modify interface to query parameters. Close #62. Close #63
closed 2019-12-24 by fhessel