fhessel/esp32_https_server #67: Is it allowable to change an SSLCert between HTTPSServer construction and calling start()?

jackjansen, 2019-12-22:
**Describe Your Goal**
I'm creating a wrapper around esp32_https_server to that has the same API as WebServer.h and WebServerSecure.h.

**What Does Your Project Look Like**

The WebServerSecure.h sequence of doing things is:

```
  WebServer server();
  server.getServer().setServerKeyAndCert(....);
  server.begin();
```

`HTTPSServer` uses a different paradigm, where the `SSLCert` is passed in during construction.
I could of course delay the construction of the `HTTPSServer`, but from inspection of the code it seems that the `SSLCert` passed in during construction isn't actually used until `start()` is called.

So, I could pass in an empty `SSLCert` to the constructor, and then fill in the key/cert with `SSLCert::setCert()` and `SSLCert::setPK()` before calling `HTTPSServer::start()`.

But it feels a bit like a hack to do this, so therefore my question: is it okay to depend on the ability to change the SSLCert between calling the HTTPSServer constructor and the start() method?

fhessel, 2019-12-22:
Do you have an example for the ESP32? For `setServerKeyAndCert()`, I can only find a related ESP8266 implementation.

Using an empty `SSLCert` to create the server instance should be fine, but changing the certificate while the server socket is up has to be prevented. I don't see a reason why the certificate should be used in the meantime, putting it into the constructor's parameter list was just a way to avoid an additional call to a setter.

It would also be important to compare the memory management of both implementations. This library takes pointers and relies on the user to make sure that the data remains valid. From [this function](https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266WiFi/src/WiFiServerSecureBearSSL.cpp#L107) and [this implementation of `X509List`](https://github.com/esp8266/Arduino/blob/e201f614e86d0f3040244aa7819036036c4f617d/libraries/ESP8266WiFi/src/BearSSLHelpers.cpp), I'd assume that the original WebServer makes a copy of the certificate and private key data. So in my opinion, that should also go into the wrapper so that it can be called with the same assumptions.

jackjansen, 2019-12-23:
Hmm, good point. I will check. I was indeed modelling after the esp8266 server (there isn't any esp32 server, that's the whole point of doing the compatibility layer:-).

I really wound't want to make a copy, these things are big, I'll see what sort of use cases I can find of whether people actually use the library without non-persistent arguments.

fhessel, 2019-12-24:
I used [WebServer.h in the Arduino ESP32 repository](https://github.com/espressif/arduino-esp32/blob/master/libraries/WebServer/src/WebServer.h) as a starting point for creating the [compat repo](https://github.com/fhessel/esp32_https_server_compat), so maybe there are also differences in the interfaces of that one and the ESP8266 WebServer.h.

I agree that the certificates and keys should only be copied if necessary due to their size. It just has to match the programmer's expectations, and that has to be clarified in the API documentation. I didn't find an explicit statement on that topic by the authors of the original WebServer.h, though, which is why I started digging in the code.