fhessel/esp32_https_server #73: Headers are case-sensitive

fhessel, 2020-03-02:
**Describe the bug**
Headers are case-sensitive, but they shouldn't be.

**Additional context**
- [RFC 2616: Hyptertext Transfer Protocol](https://tools.ietf.org/html/rfc2616#section-4.2) on HTTP headers
- [RFC 822: Standard for the format of ARPA internet messages](https://tools.ietf.org/html/rfc822#section-3.4.7) on header structure, as referenced by RFC2616
- [code that makes the case-sensitive comparison](https://github.com/fhessel/esp32_https_server/blob/9ba205ea4b0e4994ff0611e358268cb2cbdd4fd5/src/HTTPHeaders.cpp#L24-L43)

f34rdotcom, 2020-03-02:
Not sure if a better place exists to put touppper. Was trying to find std::toupper but I was unsuccessful. Hope this helps in any case. It preserves the original case of the header and does all comparisons in upper case.

```
diff --git a/src/HTTPHeaders.cpp b/src/HTTPHeaders.cpp
index 26dda1a..1dff88c 100644
--- a/src/HTTPHeaders.cpp
+++ b/src/HTTPHeaders.cpp
@@ -1,5 +1,13 @@
 #include "HTTPHeaders.hpp"
 
+std::string toupper(const std::string & s)
+{
+    std::string ret(s.size(), char());
+    for(unsigned int i = 0; i < s.size(); ++i)
+        ret[i] = (s[i] <= 'z' && s[i] >= 'a') ? s[i]-('a'-'A') : s[i];
+    return ret;
+}
+
 namespace httpsserver {
 
 HTTPHeaders::HTTPHeaders() {
@@ -13,8 +21,9 @@ HTTPHeaders::~HTTPHeaders() {
 }
 
 HTTPHeader * HTTPHeaders::get(std::string const &name) {
+  std::string ucname = toupper(name);
   for(std::vector<HTTPHeader*>::iterator header = _headers->begin(); header != _headers->end(); ++header) {
-    if ((*header)->_name.compare(name)==0) {
+    if (toupper((*header)->_name).compare(ucname) == 0) {
       return (*header);
     }
   }
@@ -22,8 +31,9 @@ HTTPHeader * HTTPHeaders::get(std::string const &name) {
 }
 
 std::string HTTPHeaders::getValue(std::string const &name) {
+  std::string ucname = toupper(name);
   for(std::vector<HTTPHeader*>::iterator header = _headers->begin(); header != _headers->end(); ++header) {
-    if ((*header)->_name.compare(name)==0) {
+    if (toupper((*header)->_name).compare(name) == 0) {
       return ((*header)->_value);
     }
   }
@@ -32,8 +42,9 @@ std::string HTTPHeaders::getValue(std::string const &name) {
 
 
 void HTTPHeaders::set(HTTPHeader * header) {
+  std::string ucname = toupper(header->_name);
   for(int i = 0; i < _headers->size(); i++) {
-    if ((*_headers)[i]->_name.compare(header->_name)==0) {
+    if (toupper((*_headers)[i]->_name).compare(ucname) == 0) {
       delete (*_headers)[i];
       (*_headers)[i] = header;
       return;

```

fhessel, 2020-03-23:
I solved it a bit differently, but could you check whether #77 works for you?

PR #77 (2020-04-04): Normalize Case of Header Names
With this PR, case of header names is normalized so that the first letter and each first letter after a non-alphanumeric character is converted to uppercase and all other letters to lowercase.

Examples:
- `content-length` becomes `Content-Length`
- `HOST` becomes `Host`

That allows working with headers independently of their case, as specified in the RFCs listed below.

**References**

- [RFC 2616: Hyptertext Transfer Protocol](https://tools.ietf.org/html/rfc2616#section-4.2) on HTTP headers
- [RFC 822: Standard for the format of ARPA internet messages](https:/ [... cut]
commit c221d22c (2020-04-04): Normalize case of header names. Resolve #73
closed 2020-04-04 by fhessel