fhessel/esp32_https_server #74: esp32_https_server with ethernet (W5500)

Unknown-0, 2020-03-05:
**Describe Your Goal**
I want to make a htt^s server other ethernet withe a w5500 shield

**What Does Your Project Look Like**
i already use W5500 for other project with the SPI and the Ethernet library and my esp32
I try to use esp32_https_server with the ethernet library but with no result can you help me?

fhessel, 2020-03-05:
I don't have that module at hand, but from what I found when looking for how to use it, it seems like it handles the Ethernet PHY and MAC layer and is connected to the ESP32 (or other microcontrollers) via SPI, and then you use for example [this library](https://github.com/PaulStoffregen/Ethernet) (and its `Ethernet` class) to talk to the module?

Then you won't be able to use it with this library. Under the hood, the library is directly talking to the lwip network stack of the [esp-idf](https://github.com/espressif/esp-idf). By default, that only works
- for WiFi in AP and Station mode
- for the Ethernet MAC which is integrated in the ESP32 with an external PHY chip (see Section 4.1.7 and Appendix A.3 in the [ESP32 Datasheet](https://www.espressif.com/sites/default/files/documentation/esp32_datasheet_en.pdf))

Or you need a library that adds your W5500 as a normal network interface in the underlying FreeRTOS. Without having looked into it, that may not be trivial from the Arduino environment.

Unknown-0, 2020-03-05:
thanks for the reply i will continue my search

fhessel, 2020-03-07:
I'll close this issue, as it isn't actionable for me. If you need any further assistance, just open another one.

closed 2020-03-07 by fhessel

ShahoffAndrew, 2020-10-26:
> thanks for the reply i will continue my search

Did you find any solution ?