fhessel/esp32_https_server #75: Need ability to force connections to close bypass keepalive.

f34rdotcom, 2020-03-07:
In some cases it is preferred to force connections to close. Sending a "Connection: close" header will force the client to close the connection even if keep alive is requested. The current code does not allow to set the Connection header as it will be replaced depending on client request for keep alive.

I need a way to force state machine to close the connection so I can reserve my limited connections and avoid this problem. 
[HTTPS:E] Could not accept() new connection

fhessel, 2020-03-07:
That's already part of #52. But as that PR combines many features and changes, I'm not really able to merge it right now.

I think to solve the root issue (new connections not being accepted), we should set two measures into place:
- Not ignoring `Connection: close` when set by the application developer (assuming he/she has a reason)
- If all connection slots are currently used, and a connection comes in, then use a least-recently-used mechanism to close one of the existing connections in favor of the new one (that's what is implemented in the aforementioned PR)

I'll take care of implementing the first one. For the latter, I've still some concerns that a malicious client can repeatedly request connections and force the ESP into dropping all existing connections with legitimate clients. Maybe the decision has to be made not only LRU-based but also w.r.t. the client IP.

fhessel, 2020-03-23:
PR #78 will handle the first problem, I'll merge it as soon as the dependent PRs are merged, too. Issue #79 tracks the implementation LRU connection handling.

PR #78 (2020-04-04): Allow Overriding Default Keep-Alive Behavior
With this PR, the user of the library can call `res->setHeader("Connection", "close");` in a handler function to circumvent the default keep-alive behavior and force the server into closing the connection, even if the client requests to keep it alive.

**Issues/PRs**

- Resolve #75
- Depends on #77
commit 50c0f590 (2020-04-04): Allow to set connection: close to disable keep-alive. Resolve #75
closed 2020-04-04 by fhessel