fhessel/esp32_https_server #76: access esp32 from secure website over internet without opening server page in separate tab first

kannieWeesNie, 2020-03-19:
i want to access my esp32 from a secure website over the internet. 

1. is it possible to access the esp from a secure website without first opening the esp32's web server page in a separate browser tab? how?

2. is it possible to access the esp32 if it is not connected to a pc? how?

thanks

i am using the esp32_https_server library. 

currently the esp32 is connected to my pc. i can successful get data from the esp by executing XMLHttpRequest() from a secure website but only if i first open the esp32 server page in a separate browser tab and by accepting the "I understand the risks and wish to continue" so that the esp page is displayed. close this page and the XMLHttpRequest() gives 499 errors.

i have published a question in stackoverflow with more details about my project but the gist of it is as described above (see [https://stackoverflow.com/questions/60759794/esp32-esp32-https-server-library-self-signed-certificate-cors-and-499-status](url))

fhessel, 2020-03-19:
First of all, from your question on Stack Overflow:

> `Status Code: 499 Request has been forbidden by antivirus`

That seems to be a problem of your Antivirus Software/Plugin/... – if there is software messing around  randomly with your connections, I cannot do much about it.

**Other than that:**

> is it possible to access the esp from a secure website without first opening the esp32's web server page in a separate browser tab?

You need a certificate that is trusted by your browser. Otherwise you will get a "network error" in the developer console of most browsers until you manually add the exception. If your ESP32 isn't exposed to the Internet and only used locally, you usually want a certificate for a local domain name, something like `myesp.local` or `myesp.internal.mycompany.com`. To get that, you must create your own certificate authority and add the corresponding CA certificate to any browser that should connect to the ESP32. Then, you can use the CA to issue a certificate for the ESP32 for the given domain name and use that certificate in your code. As the certificate is signed by a CA known to the browser, the browser will accept it without an exception (and you get that green lock icon in the address bar by default). Then, you can access the ESP32 in an XHR without any prior measures.

An easy way to setup a custom CA is [OpenVPN's easy-rsa](https://github.com/OpenVPN/easy-rsa).

These is no way to get that working for a self-signed certificate.

> is it possible to access the esp32 if it is not connected to a pc?

I don't see why that should be a problem. If your ESP32 knows how to connect to a local WiFi, just use a wall adapter to power it up and it should connect. Then, you can access the server like you did before. The serial monitor should be required only for reading the logs.

squonk11, 2020-03-19:
I am not sure but I could imagine that the question was targeting another issue: having some webpage (not on ESP32 but some other webserver) and accessing from this website via XMLHttpRequest() data from the ESP32 - @kannieWeesNie is that what you intended to do?
If yes, maybe you can read this: [Cross-Origin Resource Sharing](https://developer.mozilla.org/de/docs/Web/HTTP/CORS). This scenario can be achieved by adding the header "Access-Control-Allow-Origin: *" to the ESP32 response. 
If your scenario is not like this --> neglect my comment.

fhessel, 2020-03-19:
That could indeed be the next problem, but if the symptoms are: "it doesn't work from the start, but it works after opening the ESP's address in a separate tab", it usually is an issue of missing trust.

If your use case allows it, you could also deliver the whole website from SPIFFS (or use CDNs for huge JS frameworks), so that the user must directly access the interface and acknowledge the exception. The [REST-API](https://github.com/fhessel/esp32_https_server/tree/master/examples/REST-API) example does this.

kannieWeesNie, 2020-03-23:
thanks for your reply (and for your great library!).

is it possible that "Status Code: 499 Request has been forbidden by anti virus" is due to the self-signed certificate not being recognized by the anti-virus program? 

(if i try open the device's url in a tab, i get the page where it states that it is unsafe to proceed, and if i proceed, the device's homepage opens up but the browser indicates an unsecured connection - whereas calling the device's url from my webpage, it does not open the tab but returns the 499 status code.)

"is it possible to access the esp32 if it is not connected to a pc?... just use a wall adapter to power it up and it should connect" - yes i realized that but the local ip address (http://192.168.0.102/) won't work outside the local area network. i managed to sort this out by using port forwarding (but heard that there could be security issues when doing this). also, there is the issue of the ip address changing (the router ip address is not static) - so i tried a dynamic dns service to take care of that and managed to get that sorted out using no-ip.

my last huddle now is getting a CA certificate for the esp32 so i can access it securely.

as you can probably tell, i'm rather new at using the esp32 and especially when it comes to secure connections. 

my goal is to use the esp32 as a server and interact with my website and with various aws services. so far i can securely communicate from the esp32 to my website but not the other way around because of the lack of a certificate.

i have registered a domain name in aws route 53 and directed it to the esp32. it works fine (unsecured). i also tried using an aws certificate and cloudfront to access the esp32 which works but again not securely because of the lack of the certificate (seems aws certificate won't work in this configuration). 

so it all rests on getting a CA certificate for the esp32. i am really struggling with this and would appreciate some help as to how to do this. i lack coding experience and am daunted by all the info available online, a lot is outdated and some takes me hours to wade through only to find that it does not apply to my case. so if you could point me in the right direction, perhaps to a tutorial or youtube video, that would be most appreciated.

i came across espressif esp-id where they show how to get a CA certificate (https://docs.espressif.com/projects/esp-jumpstart/en/latest/security.html) but this seems like a whole new development environment which i am hesitant to get involved with as i don't know if it will solve my problem. 

i use the arduino core and platformio - i find there are many examples, tutorials and libraries for this. i know nothing about esp-idf, should i go this route?

Once again, thanks for your help!

fhessel, 2020-03-23:
> is it possible that "Status Code: 499 Request has been forbidden by anti virus" is due to the self-signed certificate not being recognized by the anti-virus program?

That could be the case. I cannot say for sure as I probably don't know what your software is doing in the background. It's more or less the same problem as within the browser: You need to have a trust anchor like the CA that is known to your anti-virus and/or browser and can verify the validity of the certificate presented by the ESP webserver.

> i managed to sort this out by using port forwarding (but heard that there could be security issues when doing this)

Besides security issues in your code (or, of course, this library), you will face other problems when exposing the ESP32 to the Internet. There are people running crawlers and port scanners etc., especially on the common ports like 443 for HTTPS. These tools often generate invalid packets or bursts of traffic which the ESP's network stack cannot handle properly. This was an issue in the past (see, for example, #20), but maybe it has become more stable over time. Just be aware that anyone knowing the IP can run a denial-of-service attack against the webserver by opening a small number of parallel connections. So do not use this setup if availability is critical for your application. There's just no effective measure that a MCU with some kBs of RAM can put in place.

> my last huddle now is getting a CA certificate for the esp32 so i can access it securely.

> i have registered a domain name in aws route 53 and directed it to the esp32

If that is the case, you could issue a certificate from [Let's Encrypt](https://letsencrypt.org/). That will also solve the issue of missing trust in browsers of AV software. However, I'm not aware of an implementation of the ACME protocol for certificate generation that runs on the ESP32, so creating and renewing the certificate will not be that easy. You could, for example, setup an nginx webserver first, and forward the port temporarily to that webserver. Then, create the certificates using certbot. Certbot stores the private key and certificate on your hard disk, and from there you could move them to the ESP32. However, Let's Encrypt's certificates are only valid for three month, so then you'd have to repeat that process. That's only a very rough draft of what I would try, I don't know about your experience with managing server software etc.

> i came across espressif esp-id where they show how to get a CA certificate (https://docs.espressif.com/projects/esp-jumpstart/en/latest/security.html)

What they show there is only how you would get an already issued certificate from a webserver. Let's say you want to download GitHub's certificate, you would run

```bash
$ openssl s_client -showcerts -connect github.com:443 < /dev/null
```

However, that would not get you a *new* certificate.

> this seems like a whole new development environment [...] i know nothing about esp-idf, should i go this route?

Yes, it is. At least to some degree. To put it simple, the Arduino Core for the ESP32 is an application written for the ESP-IDF that allows you to insert a `setup` and a `main` function just like you would do with Arduino. The complexity of the ESP-IDF are hidden from you. But this is also the case for many features. If you want to have full control over your ESP32, you'd need to go for the ESP-IDF, but I would not suggest it if you're a beginner.

kannieWeesNie, 2020-03-29:
thanks, once again for your help.

i managed to get an SSL certificate but after loading it up and running the app, i get this message: setupCert failed

i'd appreciate it if could help me overcome this issue. here's what i have done:

i have a CA certificate which i downloaded into these files:

ca_bundle.crt
certificate.crt
private.key  

and assumed they are all in PEM format, as they all display as...-----BEGIN PRIVATE KEY-----etc.

i then converted them to DER format using openSSl. i then uploaded the certificate and private key files (DER formats) to the SPIFFS using pio run -t uploadfs.

running the app, i get the following in the serial monitor:

SPIFFS has been mounted. 
getting certificate...   
Reading certificate from SPIFFS.
Read 1371 bytes of certificate and 294 bytes of key from SPIFFS
we have a certificate....
Connecting to WiFi..
192.168.0.102
setupCert failed

here's the relevant part (the getCertificate()) of the code i use to get the certificate from the SPIFFS:

File keyFile = SPIFFS.open("/private.der");
File certFile = SPIFFS.open("/certificate.der");
	
	
{
        Serial.println("Reading certificate from SPIFFS.");

        // The files exist, so we can create a certificate based on them
        size_t keySize = keyFile.size();
        size_t certSize = certFile.size();

        uint8_t *keyBuffer = new uint8_t[keySize];
        if (keyBuffer == NULL)
        {
            Serial.println("Not enough memory to load privat key");
            return NULL;
        }

        uint8_t *certBuffer = new uint8_t[certSize];
        if (certBuffer == NULL)
        {
            delete[] keyBuffer;
            Serial.println("Not enough memory to load certificate");
            return NULL;
        }
        
        keyFile.read(keyBuffer, keySize);
        certFile.read(certBuffer, certSize);

        // Close the files
        keyFile.close();
        certFile.close();
        Serial.printf("Read %u bytes of certificate and %u bytes of key from SPIFFS\n", certSize, keySize);
        return new SSLCert(certBuffer, certSize, keyBuffer, keySize);
}

fhessel, 2020-03-29:
294 bytes sounds rather small.

Could you run openssl again to verify that the files are valid DER files?

```bash
openssl pkey -inform der -in private.der -text
```

Output should be something like:

```
-----BEGIN PRIVATE KEY-----
<base64-output>
-----END PRIVATE KEY-----
RSA Private-Key: (1024 bit, 2 primes)
modulus:
    <hex-output>
publicExponent: 65537 (0x10001)
privateExponent:
    <hex-output>
prime1:
    <hex-output>
prime2:
    <hex-output>
exponent1:
    <hex-output>
exponent2:
    <hex-output>
coefficient:
    <hex-output>
```

And the same for the certificate file:

```bash
openssl x509 -inform DER -in certificate.der -text
```

That should give you something like:

```
Certificate:
    Data:
        Version: 1 (0x0)
        Serial Number: 2 (0x2)
        Signature Algorithm: sha256WithRSAEncryption
        Issuer: C = DE, ST = BE, L = Berlin, O = MyCompany, CN = myca.local
        Validity
            Not Before: Mar 17 21:07:09 2020 GMT
            Not After : Mar 15 21:07:09 2030 GMT
        Subject: C = DE, ST = BE, L = Berlin, O = MyCompany, CN = esp32.local
        Subject Public Key Info:
            Public Key Algorithm: rsaEncryption
                RSA Public-Key: (1024 bit)
                Modulus:
                    <hex-output>
                Exponent: 65537 (0x10001)
    Signature Algorithm: sha256WithRSAEncryption
         <hex-output>
-----BEGIN CERTIFICATE-----
<base64-output>
-----END CERTIFICATE-----
```

What commands did you use to convert the certificates?

kannieWeesNie, 2020-04-04:
Thank you very much! got it working. as you pointed out, i had incorrectly converted the certificate and private key. working perfectly after converting them correctly. thanks very much for your help. much appreciated!

closed 2020-04-04 by kannieWeesNie