fhessel/esp32_https_server #84: Parsing Post Requests twice fails (HTTPURLEncodedBodyParser)

JAY-M-ERICSSON, 2020-04-14:
I'm trying to use your library (https) and my web site uses POST for all requests.

A POST always include a session UUID that lets me identify which session is making a request, which I use to establish a server side context in which my call back will be executed.

So When I get an incoming request, I call a function which creates a parser and walk through the request to find my UUID entry. I compare it with known UUIDs, setup a context and return. 

Then the callback proceeds and in some cases I need to parse again the POST data to extract other elements of the request.

--> that's where I'm having an issue, it seems that creating a new parser on the same request fails, there is no more attributes to read.

Is that intended behavior ? does parsing the request once kills the post data ???

thx

(as a side note, it would be great to not instantiate so many strings and give us const pointers into the body to extract the data we are interested in).

fhessel, 2020-04-14:
> Is that intended behavior ? does parsing the request once kills the post data?

Yes, it is. The BodyParsers *consume* the body. Otherwise you wouldn't be able to handle large requests that don't fit in memory.

If you need random field access an you know that everything will fit into memory, my suggestion would be to create a struct for all required variables and assign them while iterating through the body. Alternatively, you could use a container like `std::map`, if you don't know the number or name of keys in advance.

> (as a side note, it would be great to not instantiate so many strings and give us const pointers into the body to extract the data we are interested in).

For the same reason as above, there is no time at which the whole body is guaranteed to be present in memory. If I would implement such an interface now, that implementation would break as soon as the library can handle different content encoding for request bodies, where all input needs to be converted on the fly. My focus is more on reducing redundant buffers on different layers of processing.

JAY-M-ERICSSON, 2020-04-14:
Thanks for explaining the rationale, that makes sense.
I’ll update serialization/deserialisation of my context - maybe as a JSON and work from there 
Thx for the good work

closed 2020-04-15 by fhessel