fhessel/esp32_https_server #88: Can't get Let's Encrypt certificates to work

kaiomatico, 2020-05-02:
**Describe Your Goal**
I want to convert my valid let's encrypt certificate to the correct DER format to use it on the HTTPS Async example

**What Does Your Project Look Like**
The example HTTPS Async sketch with only the cert.h and private_key.h replaced. I got the following files from Let's Encrypt:
- cert.pem
- chain.pem
- fullchain.pem
- privkey.pem

I have tried the following commands with varying level of success to generate the example.crt.DER which then gets converted to cert.h with xxd:
- openssl x509 -in fullchain.pem -outform DER -out example.crt.DER
- openssl x509 -in cert.pem -outform DER -out example.crt.DER
- opening a properly configured Apache webserver with the certificate and then download the certificate with chrome and saving it as DER file. (I have tried it at the root level as well as only on the domain level)

I test using curl -v which always fails at :
* ALPN, offering h2
* ALPN, offering http/1.1
* successfully set certificate verify locations:
*   CAfile: /etc/ssl/certs/ca-certificates.crt
  CApath: /etc/ssl/certs
* TLSv1.3 (OUT), TLS handshake, Client hello (1):
* TLSv1.3 (IN), TLS handshake, Server hello (2):
* TLSv1.2 (IN), TLS handshake, Certificate (11):
* TLSv1.2 (OUT), TLS alert, unknown CA (560):
* SSL certificate problem: unable to get local issuer certificate
* Closing connection 0
curl: (60) SSL certificate problem: unable to get local issuer certificate
More details here: https://curl.haxx.se/docs/sslcerts.html

And using the chrome exported certificate from the root level it fails at:
* ALPN, offering h2
* ALPN, offering http/1.1
* successfully set certificate verify locations:
*   CAfile: /etc/ssl/certs/ca-certificates.crt
  CApath: /etc/ssl/certs
* TLSv1.3 (OUT), TLS handshake, Client hello (1):
* TLSv1.3 (IN), TLS handshake, Server hello (2):
* TLSv1.2 (IN), TLS handshake, Certificate (11):
* TLSv1.2 (IN), TLS handshake, Server key exchange (12):
* TLSv1.2 (OUT), TLS alert, decrypt error (563):
* error:0407008A:rsa routines:RSA_padding_check_PKCS1_type_1:invalid padding
* Closing connection 0
curl: (35) error:0407008A:rsa routines:RSA_padding_check_PKCS1_type_1:invalid padding

So how do I convert let's encrypt certificated properly?

**ESP32 Module**
ESP32 Dev Module breakout board, no RAM addon or anything fancy

**Software (please complete the following information if applicable)**
 - IDE and Version:  Arduino 1.8.12
 - OS: Ubuntu 20.0.4
 - Client used to access the server: curl 7.68.0

fhessel, 2020-06-06:
Sorry for the late reply. If I remember correctly, using the full chain will not work, only the last certificate that is the one you want to use with the server, and for which you have the matching private key.

So this should be the correct syntax if you have a `cert.pem` file:

```bash
# check the certificate, only a single certificate should be shown (not a chain):
openssl x509 -inform pem -in cert.pem -text
# Convert the certificate
openssl x509 -in cert.pem -outform DER -out cert.der
# Check the converted certificate (output should be the same as for the first command)
openssl x509 -inform der -in cert.der -text
```

For the private key (in .pem format) to the following. Make sure you actually have an RSA key. EC won't work with the server.

```bash
openssl rsa -in cert.key -outform der -out cert.key.der
```

As a last step, you can verify that your private key and certificate in the DER files do match:

```bash
# Get modulus of the key
openssl rsa -modulus -noout -in cert.key.der -inform der | openssl sha256
# Get modulus of the cert
openssl x509 -modulus -noout -in cert.der -inform der | openssl sha256
```

If both commands show the same output, the files should be okay and that should not be the problem.

What might be an issue though is if the intermediate CA used by Let's Encrypt isn't known to your browser (or curl, respectively). In that case, you would have a working certificate presented by the ESP32, but the client isn't able to verify it. These lines from your test with curl look like that could indeed be the problem:

```
TLSv1.2 (OUT), TLS alert, unknown CA (560):
SSL certificate problem: unable to get local issuer certificate
```

Does `curl --insecure [...]` solve the problem? In that case, curl would ignore being not able to validate the certificate, but at least we would see that the certificate as such was deployed correctly. In that case, I'd convert this issue into a feature request to support certificate chains delivered by the server.

kaiomatico, 2020-06-08:
Thank you for the answer, I will test it on the weekend :-)

kaiomatico, 2020-06-20:
Ok I finally got around to test it today with the commands you gave me. Sadly curl still shows the same error message. When using curl with --insecure it works.
```
* TCP_NODELAY set
* Connected to [Censored my DNS Name] port 443 (#0)
* ALPN, offering h2
* ALPN, offering http/1.1
* successfully set certificate verify locations:
*   CAfile: /etc/ssl/certs/ca-certificates.crt
  CApath: /etc/ssl/certs
* TLSv1.3 (OUT), TLS handshake, Client hello (1):
* TLSv1.3 (IN), TLS handshake, Server hello (2):
* TLSv1.2 (IN), TLS handshake, Certificate (11):
* TLSv1.2 (IN), TLS handshake, Server key exchange (12):
* TLSv1.2 (IN), TLS handshake, Server finished (14):
* TLSv1.2 (OUT), TLS handshake, Client key exchange (16):
* TLSv1.2 (OUT), TLS change cipher, Change cipher spec (1):
* TLSv1.2 (OUT), TLS handshake, Finished (20):
* TLSv1.2 (IN), TLS handshake, Finished (20):
* SSL connection using TLSv1.2 / ECDHE-RSA-AES256-GCM-SHA384
* ALPN, server did not agree to a protocol
* Server certificate:
*  subject: CN=[Censored my DNS Name]
*  start date: Jun 20 16:17:11 2020 GMT
*  expire date: Sep 18 16:17:11 2020 GMT
*  issuer: C=US; O=Let's Encrypt; CN=Let's Encrypt Authority X3
*  SSL certificate verify result: unable to get local issuer certificate (20), continuing anyway.
> GET / HTTP/1.1

```
So thank you very much for your effort :) Sad that this is probably nothing we / you could fix. So I am closing this issue now, have a nice weekend and thank you for the awesome project!

closed 2020-06-20 by kaiomatico

fhessel, 2020-06-20:
Thanks for checking that! If `curl --insecure` works but says `SSL certificate verify result: unable to get local issuer certificate (20), continuing anyway.`, that indeed sounds like a problem with the certificate _chain_, not the certificate itself. I created #93 to track work on that.