fhessel/esp32_https_server #94: Support for client certificates

gsuberland, 2020-08-02:
At current it seems that this library does not support client certificates for authentication. My understanding is that you're using mbedtls for the underlying TLS implementation, and it does support client certificates (see [mbedtls_ssl_conf_authmode](https://tls.mbed.org/api/ssl_8h.html#a5695285c9dbfefec295012b566290f37) for reference). It would be nice to see this support ported across, since it offers a much safer way of performing authenticated access than basic auth over HTTPS.

fhessel, 2020-08-02:
> It would be nice to see this support ported across, since it offers a much safer way of performing authenticated access than basic auth over HTTPS.

Full ACK, I didn't consider it as high priority as most (browser) clients don't make use of it as it's not particularly user-friendly to configure.

How would a scenario look like for you? Would you have a number of certificates + public key on your device, each for a given user, and maybe the option to add more users + certificates later on? Or would you use it in combination with a (maybe private) CA that signs user's certificates, so the ESP does only need to know a single issuer certificate?

gsuberland, 2020-08-02:
My personal use-case would be for performing authenticated pull on IoT sensor devices in an environment with an internal CA, in which there'd only be one client certificate (i.e. that of the system orchestrating pulls) configured on all of the devices.

The ability to add multiple public keys for client certs on the server would certainly be useful for other use-cases, and it seems like it shouldn't be too difficult considering the design of the API, which appears to work by calling the same function a bunch of times with different keys in order to register them, meaning that this library wouldn't need to keep track of anything on that front.

tionis, 2021-07-01:
Is it possible to achieve client certificate verification (via a private CA)  with the current code?  
If not which part would I have to modify?
I'm not very familiar with C++ and embedded systems, so any Pointers would be greatly appreciated!