fhessel/esp32_https_server #97: Captive portal example

flatsiedatsie, 2020-08-31:
**Is your feature request related to a problem? Please describe.**
Thank you for the many wonderul examples, they have helped a lot. But one thing I have not been able to get working is turning the EP32 into a hotspot that also redirects users to the website that it's hosting. I would like users to be able to connect to the ESP's wifi hotspot, and that the browser would then automatically show the website that it's hosting. No login is required.

I tried to get this to work by combining examples I found with this code. However, when I enables a local DNS host, it always resulted in 'guru meditation' error. This must be my own user error. But after trying to get it to work for a few hours, and feeling that I may not be the first to try and use an ESP32 this way, I thought I'd ask for any tips, suggestions, or perhaps even an example.

**Describe the solution you'd like**
Advice would be welcome. An example would be amazing. I would be happy to help develop an example and then share it as part of the exampes of this addon.

**Describe alternatives you've considered**
I spent a lot of time trying to make it work myself, but my skill level is probably not high enough.

**Additional context**
It's for an art project. The idea is that people who visit an exhibit can control a water fountain themselves by connecting to the device's hotspot with their mobile phone, and then automatically being guided to the website on it. Then they can control the fountain themselves.

fhessel, 2020-09-05:
Could you try #98 and check if that works for you?

I had the same issue with a **lot** of guru meditation, but that seems to be caused by the way the DHCP server is implemented in the AP mode for Arduino. It seems to store configuration in the non-volatile flash partition, which then causes the reboot shortly after a client connects. So in case you face that issue, maybe just try erasing the whole flash and then uploading a clean sketch.

Limitations of the example are described in the PR, but the basic functionality works. You might need to turn off mobile data when you're connected to the AP, I'm not sure if that can be circumvented.

flatsiedatsie, 2020-09-07:
Wow, thank you!

I've actually also managed to get it to work at the same time. The trick that worked for me was:

DNS as per the examples -> create a http server too, which captures root -> index.html on non-https server forwards to https server using a `meta` refresh element.

I'll look into your solution (it's probably less of a round-about way) and implement that instead.

flatsiedatsie, 2020-09-07:
Looking at your code it seems you did the same thing that I stubled upon: adding an insecure http server step solves the problem.

I tried running your code, but it didn't work.
- upload code
- connect to open hotspot `captiveESP`
- open random website in browser (`https://www.youtube.com/`) in Chromium on macos
- `ERR_CONNECTION_REFUSED`

This is in the serial output:
```
Setting up WiFi... OK
Starting DNS... OK
Starting HTTP server... OK
[HTTPS:I] New connection. Socket FID=56
[HTTPS:I] Request: GET /bag (FID=56)
[HTTPS:W] Could not find a matching resource
[HTTPS:I] Connection closed. Socket FID=56
[HTTPS:I] New connection. Socket FID=57
[HTTPS:I] Request: GET /bag (FID=57)
[HTTPS:W] Could not find a matching resource
[HTTPS:I] Connection closed. Socket FID=57
[HTTPS:I] New connection. Socket FID=58
[HTTPS:I] Request: GET /bag (FID=58)
[HTTPS:W] Could not find a matching resource
[HTTPS:I] Connection closed. Socket FID=58
[HTTPS:I] New connection. Socket FID=59
[HTTPS:I] Request: GET /bag (FID=59)
```

I disabled the `https-everywhere` plugin, but that didn't change things.

The server does seem to turn `http` into `https`, but it seems to do this before it moves things to the internal IP address?

A small typo in the example code:
`Do not uses connection config persistence`

Here is my working messy code:

```

// For captive portal
#include <DNSServer.h>
const byte DNS_PORT = 53;
DNSServer dnsServer;

#define USE_SD // true               // This switches between using an SD card of SPIFFS as the filesystem

#define LED 2

unsigned long last_unlock_time = 0;

#include "FS.h"

#ifdef USE_SD
#include "SD.h"
#include "SPI.h"
#define PIN_NUM_MISO 16 
#define PIN_NUM_MOSI 13 
#define PIN_NUM_SCK  14 
#define PIN_NUM_SS   15
#else
#include <SPIFFS.h>
#endif

// We will use wifi
#include <WiFi.h>

// Trying to handle captive portal crash
//wifi_init_config_t wifi_init_config = WIFI_INIT_CONFIG_DEFAULT();
//wifi_init_config.ampdu_rx_enable = 0;
[... 270 lines cut ...]
  res->println("<html>");
  res->println("<head><title>Water fountain</title></head>");
  if (req->isSecure()) {

  }else{
    res->println("<meta http-equiv=\"Refresh\" content=\"0; url='https://192.168.4.1/index.html'\" />");
  }
  res->println("<body>");
  res->println("<h1>Forwarding...</h1>");

  // You can check if you are connected over a secure connection, eg. if you
  // want to use authentication and redirect the user to a secure connection
  // for that
  if (req->isSecure()) {
    res->println("<p>You are connected via <strong>HTTPS</strong>.</p>");
  } else {
    res->println("<p>You are connected via <strong>HTTP</strong>.</p>");
  }

  res->println("</body>");
  res->println("</html>");
}

// there are some functions to handle files here, which I've removed for brevity.
```