fhessel/esp32_https_server #98: Browser complaining about the connection is not private.

zekageri, 2020-09-03:
**I want to be able to push notifications to the client and some other javascript stuffs thats not possible without https.**

I managed to run the async server example, i can reach my website by typing `https://myhome.local/ ` and i get the connection is not private on every browser. 

_I understand that the browser requesting some certificate, the server sends the generated one to it and that should be it?
Do i need to send back some other client side certificate to the server ( in js? ) to be able to open the webpage with https securely?_

fhessel, 2020-09-03:
Your problem is with trust between your browser and the server on the ESP32. Explaining the concept of public key infrastructures (PKIs) would be out of scope for this issue, but that is what you want to read about. Andreas Spiess also did [a video that explains the basics](https://www.youtube.com/watch?v=Wm1xKj4bKsY) in context of ESP8266/32.

Brief explanation on what happens:

- The server answers with its **certificate and public key** when it gets a TLS connection request
- The client (your browser) wants to verify it. The **certificate is signed** by another party and their certificate. In the Internet, that's usually a **Certificate Authority (CA)**.
- This CA also has its certificate, which again may be signed by another CA and so on, creating a **certificate chain**.
- At some point, this chain needs to end at a certificate that is **known to** and **trusted by** your browser. In the Internet, your browser comes with some **root certificates** that are trusted by default, so for websites, this works.

Now to the problem:

- The certificate on the server is signed by itself, therefore it's called a **self-signed** certificate.
- The certificate chain has a length of 1.
- None of these certificates is known to your browser.

Result:

- You get the "not private" warning, because the browser cannot **verify trust**.

Possible Solutions:

- For testing: Accept the exception in your browser, reload. Make sure to use the same certificate each time (so don't use generation of self-signed certificates on the ESP, or you need to add the exception every time the cert is re-generated).
- If you use the devices internally, in your network: Set up your own CA, and install it in all browsers or devices that need to communicate with your ESP32 webserver. Then you can issue (sign) certificates for your own IoT devices. But be aware: Everyone with access to that CA can then issue arbitrary certificates that will be trusted by your browser, so these people can run MitM attacks and read/change **all** your web traffic.
- For access from the Internet: You need to get a certificate from a proper CA. Let's encrypt is free of charge, but you need to configure a public DNS resolution properly and you'll have to refresh the certificate every three months, which afair hasn't been implemented for the ESP32 by now.

That is not a complete solution to your problem, but hopefully some pointers to read on. I'll close the issue with that, but if you have more questions, feel free to ask.

closed 2020-09-03 by fhessel

zekageri, 2020-09-04:
### Thank you so much for the explanation. 

What if i want to reach the webserver of my esp32 from not just windows/mac computers but from mobiles, tablets etc...? 

**Can i add my own CA as a trusted CA to these devices too?**
**What if i want to host my server only on local network? Does routers have some method to verify my CA to all connected browsers?**

_For example, the devices browser trying to reach an external source to verify my CA. I belive that this is happening trought my local router, so the browser first asks my router that if this CA is ok?_

If i understand you correctly, there is no way for a browser to trust in a https server out of the box. 
**That's so sad, but thank you!**

fhessel, 2020-09-04:
> What if i want to reach the webserver of my esp32 from not just windows/mac computers but from mobiles, tablets etc...?
> Can i add my own CA as a trusted CA to these devices too?

Yes! And that's what I would suggest for private projects that you use just in your own network. How it's done differs based on your device's OS, but if you search for "Install CA on Android/iOS/..." you should find a tutorial. Make sure to install the CA certificate on the devices, then all certificates that you create with that CA are automatically valid on the devices. So if you get another module, just assign a hostname, issue a new certificate and you're done. No more distribution to devices. That's exactly why there Root CAs exist for Internet scale.

> For example, the devices browser trying to reach an external source to verify my CA. I belive that this is happening trought my local router, so the browser first asks my router that if this CA is ok?

**Verifying trust** is done locally. Your browser/device has a "trust store" (name may vary) that contains all trusted certificates. Your router should not be involved in this.¹ If the browser cannot find a chain ending in its trust store, you get the warning.

> If i understand you correctly, there is no way for a browser to trust in a https server out of the box.

You would need a certificate from a CA known to your browser. As certificates are bound to a hostname, in a first step, you need a domain. Let's assume you own zekageri.xyz. Then you have two options:

1. Buy a commercial certificate from a "classic" CA. Usually too costly for a locally used project.
2. Use [Let's Encrypt](https://letsencrypt.org). They offer free certificates. However, they need to verify automatically that you own zekageri.xyz. Therefore, you would need to run a webserver there that can answer the [ACME protocol](https://en.wikipedia.org/wiki/Automated_Certificate_Management_Environment). As mentioned, I don't know of an ESP32-based tool that can do this. And your ESP32 needs to be reachable over the internet, which might be a problem on its own on many consumer-grade connections (in particular, as the ESP32 has no full IPv6 support under Arduino).

For these reasons I would strongly suggest to read a bit up on how to setup a personal CA. That also helps great in understanding how all this stuff goes together.

¹ There is an exception to this: Some anti virus tools or company-level firewalls think it is a good idea to break every TLS connection to examine the traffic. For that, they would install a CA certificate in your browser and then run the mentioned MitM attack against you. But you should know if that's the case for you.

zekageri, 2020-09-09:
Oh god, you are a gold man. Really thank you for all of that information. From now on, i will go and read about everything that you suggested.