fhessel/esp32_https_server #101: How to use default GET, DELETE and POST handler simultaneously?

marciot, 2020-10-30:
**Describe Your Goal**

I am trying to implement an interface to manage files on the server through GET, DELETE and POST requests.

In other HTTP libraries I've used this is accomplished by creating a catch all (i.e. default handler) that performs the appropriate function on the resource indicated by the URL depending on the HTTP method. However, since in this library there appears to only one default handler, I can only implement one of the three operations at a time.

If there were a "addDefaultNode", as opposed to a "setDefaultNode", I would write my code like this:

```
void handleGetFile(HTTPRequest * req, HTTPResponse * res) {
    std::string filename = req->getRequestString();
    ...
    // Serve filename
}

void handleUploadFile(HTTPRequest * req, HTTPResponse * res) {
    std::string filename = req->getRequestString();
    ...
    // Upload filename
}

void handleDelete(HTTPRequest * req, HTTPResponse * res) {
    std::string filename = req->getRequestString();
    ...
    // Delete filename
}

void startWebServer() {
    webServer.addDefaultNode(new ResourceNode("", "GET",    &handleGetFile));
    webServer.addDefaultNode(new ResourceNode("", "DELETE", &handleDeleteFile));
    webServer.addDefaultNode(new ResourceNode("", "POST",   &handleUploadFile));
    webServer.start();
}
```

There are alternative ways in which this could work. For example, maybe a single handler could support all methods, but this would require the ability to ResourceNodes that took wildcard methods. something like this:

```
void handleFileOperations(HTTPRequest * req, HTTPResponse * res) {
    std::string filename = req->getRequestString();
    std::string method = req->getMethod();
    if(method == "GET") {
       // Serve filename
    }
    else if(method == "POST") {
       // Upload filename
    }
    else if(method == "DELETE") {
       // Delete filename
    }
}

void startWebServer() {
    webServer.setDefaultNode(new ResourceNode("", "*", &handleFileOperations));
    webServer.start();
}
```

I can't find any documentation that indicates whether this functionality already exists, perhaps in some manner I have not yet described.

Any help would be appreciated!

fhessel, 2020-10-31:
Both approaches are pretty close to how I'd do it.

**Based on your first example:** If you want to use the same handler for multiple methods, you can just register it multiple times. If we take a function `handleFileOperations(HTTPRequest, HTTPResponse)`, the following setup code would be totally fine:

```c++
void startWebServer() {
    webServer.registerNode(new ResourceNode("/*", "GET", &handleFileOperations));
    webServer.registerNode(new ResourceNode("/*", "POST", &handleFileOperations));
    webServer.registerNode(new ResourceNode("/*", "DELETE", &handleFileOperations));
    webServer.start();
}
```

Method wildcards, however, are not yet supported. On a positive side, you know that the method is only called with valid and expected methods.

Also, be aware that using `""` for the first parameter will only match `http://yourserver/` but not any additional path. You can use `/*` as shown in the [parameters example](https://github.com/fhessel/esp32_https_server/blob/daf6ca1c4359e9c9fdea561a748ebc8be029fb23/examples/Parameters/Parameters.ino#L89) or the [REST API example](https://github.com/fhessel/esp32_https_server/blob/daf6ca1c4359e9c9fdea561a748ebc8be029fb23/examples/REST-API/REST-API.ino#L181). But `/*` only matches a single level in the path hierarchy. It will match `http:///myfile.txt` but not `http:///files/somefile.txt`.

**Extending your second example:** `HTTPServer::setDefaultNode(ResourceNode)` can be used to register a handler function that will be called if _no other node_ matches, independently of the _method or path_ used. Usually, you would use that to define a 404 page (like in the [static page example](https://github.com/fhessel/esp32_https_server/blob/daf6ca1c4359e9c9fdea561a748ebc8be029fb23/examples/Static-Page/Static-Page.ino#L86)), but you can also use it as a file browser and editor. `HTTPRequest::getRequestString()` should give you the full request string, but you're on your own with parsing then.

marciot, 2020-11-02:
@fhessel: Since I would like to be able to have files be GET/POST/DELETED from any directory, I think the second example is the best. I was not aware that the default handler could have a blank method as well as a blank path. Given that is the case, I think all I need to do is call `req->getMethod()` to figure out which function to perform on the resource.

I think that is exactly what I was looking for. Thank you for pointing me in the right direction!

closed 2020-11-02 by marciot