fhessel/esp32_https_server #102: Can't import and use your library in another library :) Why? If it is .h file, there is no problem but .hpp seems can't be used

robertgregor, 2020-11-06:
I have another library and want to use your library in my library. But seems, it is not possible:

C:\RH\Sources\RH\RemoteHomeArduinoSketches\WiFi_ESP32\libraries\RemoteHome\src/RemoteHome.h:32:5: error: 'HTTPSServer' does not name a type
     HTTPSServer * secureServer;
     ^
C:\RH\Sources\RH\RemoteHomeArduinoSketches\WiFi_ESP32\libraries\RemoteHome\src/RemoteHome.h:33:5: error: 'HTTPServer' does not name a type
     HTTPServer * server;
     ^
C:\RH\Sources\RH\RemoteHomeArduinoSketches\WiFi_ESP32\libraries\RemoteHome\src/RemoteHome.h:129:23: error: 'HTTPRequest' has not been declared
     void handleSPIFFS(HTTPRequest * req, HTTPResponse * res);
                       ^
C:\RH\Sources\RH\RemoteHomeArduinoSketches\WiFi_ESP32\libraries\RemoteHome\src/RemoteHome.h:129:42: error: 'HTTPResponse' has not been declared
     void handleSPIFFS(HTTPRequest * req, HTTPResponse * res);

Do you have an idea why?

fhessel, 2020-11-06:
Did you [use the namespace](https://github.com/fhessel/esp32_https_server#get-started)? After the imports, write:

```c++
// Easier access to the classes of the server
using namespace httpsserver;
```

Otherwise, you'd need to prefix every class name with it, like this:

```c++
httpsserver::HTTPServer * server;

void handleSPIFFS(httpsserver::HTTPRequest *req, HTTPResponse *res) {
   // ...
}
```

I see that it's an easy pitfall, but it allows isolating the class names and preventing collisions with other libraries.

robertgregor, 2020-11-08:
I see, yes, I am stupid. Thanks! I didn't spot that, that you use it 👍

closed 2020-11-08 by robertgregor