fhessel/esp32_https_server #104: How to parse GET parameters

dvdbrk, 2020-11-18:
**Describe Your Goal**
I am trying to get query string to show in Serial console / make a `char` variable out of it.
I think this is something pretty simple and C++ related, but as a total beginer, this is causing me hours of headache. Please help.

**What Does Your Project Look Like**
I am using `Parameters example` code.

```c++

ResourceParameters *params = req->getParams();
  for(auto it = params->beginQueryParameters(); it != params->endQueryParameters(); ++it) {
    res->print("<tr><td>");
    
    // The iterator yields std::pairs of std::strings. The first value contains the parameter key
    res->printStd((*it).first);
    res->print("</td><td>");

    //  and the second value contains the parameter value
    res->printStd((*it).second);
    res->println("</td></tr>");

    // this is the problem
    // I am hoping to get a parameter name here (i.e. '&user=', '&test=', without & sign)
    // none of the below lines work
    const char* stringValue = (*it).first; 
    Serial.println(res->printStd((*it).first));
    
  }
  res->println("</table>");
```

fhessel, 2020-11-18:
```c++
Serial.println((*it).first.c_str());
```

should do the trick.

dvdbrk, 2020-11-18:
Works like a charm, thank you!

closed 2020-11-18 by dvdbrk