fhessel/esp32_https_server #105: How to pass a object member function to ResourceNode constructor?

marfanjr, 2020-12-06:
**Describe Your Goal**
What are you trying to achieve?

**What Does Your Project Look Like**
Provide a description of your project structure, other libraries that you use, ...

If you can boil your question down to a [minimal working example](https://en.wikipedia.org/wiki/Minimal_working_example), please do so and share the code here:

```c++
// some code
```

**ESP32 Module**
Please provide specifications of your module
- RAM/PSRAM: 
- Flash Size: 
- Other special properties: 

**Software (please complete the following information if applicable)**
 - IDE and Version: [e.g. Arduino 1.8.7, Sloeber 4.3, PlatformIO 3.6.2]
 - OS: [e.g. Debian stretch]
 - Client used to access the server: [e.g. Chrome 70.0, curl 7.55.1 - if possible, please try to reproduce the bug with a tool like curl -v to have most control on what's happening]

**Additional context**
Add any other context about the problem here.

fhessel, 2020-12-07:
That won't work with the current release version, but #91 enables it when merged.

However, you cannot use your class method without an instance. So at some point, you need to create an instance of your HttpsHandler.

If that instance is called `myHandler`, the code would look like this:

```c++
// At the top of your sketch:
#include <HTTPSCallbackFunction.hpp>

// In your setup code:
HTTPSCallbackFunction handler =
          std::bind(&HttpsHandler::root, &myHandler, std::placeholders::_1, std::placeholders::_2);
server.registerNode(new ResourceNode("/", "GET", handler));
```

A bit simplified, using `bind` creates a pointer to that root method that knows on which instance it should be executed. The placeholders allow to pass-through the `req` and `res` parameters.

#90 has more details on this.

marfanjr, 2020-12-07:
Thanks for the quick response!
Do you have any predictions of when the PR will be merged?
I tested the PR branch and it worked perfectly!

fhessel, 2020-12-07:
Thanks for testing and giving feedback! I'm currently working on building a reliable, automated, and hardware-based CI pipeline for this repository, since I want to be sure that after merging a PR the examples still work (not only that they are able to build) and to be able to test all functionality in isolation. Doing that manually for each PR takes me some hours each time I'm merging something, which makes it hard to merge even little changes.

For that reason I decided to prioritize working on the CI first instead of spending my (currently quite limited) spare time on manual testing – that's also the reason why most issues are stale at the moment. I hope to get everything running around Christmas, and then merging PRs and working on issues should be easier and be doable with more confidence in the codebase.

If I don't stumble upon any issues with the way the PR is implemented, it will stay as it is. So you could in theory start working on the PR branch, since the only code changes to 1.0.0 are the ones required for implementing the callbacks, all other commits are just documentation, and when 1.1.0 is out, you could switch back to using normal semver.

marfanjr, 2020-12-08:
Excellent decision! I will continue using the branch pr and awaiting the merger.

Thank you very much for your attention! I hope to be able to contribute to the community as I progress in my studies.