fhessel/esp32_https_server #108: [HTTPS:W] Header length exceeded.

robertgregor, 2020-12-18:
Hi, I am getting this error, when I submit seems long GET request with a lot of parameters:
i.e.:
Request URL: http://192.168.1.104/deviceconfig?timeZone=1&dl=Off&batop=Off&mdns=Testmdns&notifperiod=600&moduleid=123456&loglevel=3&logdest=0&enckey=&outcounter=0&incounter=0&_=1608300579947
Referrer Policy: strict-origin-when-cross-origin
Accept: */*
Accept-Encoding: gzip, deflate
Accept-Language: en-US,en;q=0.9,cs-CZ;q=0.8,cs;q=0.7
Cache-Control: no-cache
Connection: keep-alive
Host: 192.168.1.104
Pragma: no-cache
Referer: http://192.168.1.104/
User-Agent: Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/87.0.4280.88 Safari/537.36
X-Requested-With: XMLHttpRequest

Do you have an idea, why I have such behavior? Is it because of long GET request??? Because when I am using the request like that i.e. it works. And headers are the same:
Request URL: http://192.168.1.104/loadmainpreferences?_=1608300580006
Request Method: GET
Status Code: 200 OK
Remote Address: 192.168.1.104:80
Referrer Policy: strict-origin-when-cross-origin
Connection: keep-alive
Content-Length: 363
Content-Type: application/json
Accept: application/json, text/javascript, */*; q=0.01
Accept-Encoding: gzip, deflate
Accept-Language: en-US,en;q=0.9,cs-CZ;q=0.8,cs;q=0.7
Cache-Control: no-cache
Connection: keep-alive
Host: 192.168.1.104
Pragma: no-cache
Referer: http://192.168.1.104/
User-Agent: Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/87.0.4280.88 Safari/537.36
X-Requested-With: XMLHttpRequest

fhessel, 2020-12-18:
The limit is configured [here](https://github.com/fhessel/esp32_https_server/blob/daf6ca1c4359e9c9fdea561a748ebc8be029fb23/src/HTTPSServerConstants.hpp#L54). Allowing for arbitrary lengths would directly lead to a memory exhaustion vulnerability, so there are hard limits.

If you're using PlatformIO, you can override the configuration in your platformio.ini file to a value that covers your longest request strings:
```ini
build_flags += -DHTTPS_REQUEST_MAX_REQUEST_LENGTH=512
```

robertgregor, 2020-12-18:
I see. So the issue it the lenhth of the GET request. So probably the error shall be, that no header length exceeded, but the get request exceeded.