fhessel/esp32_https_server #109: Endless loop on HTTPConnection::readLine

erosinnocenti, 2021-01-13:
Hello,
I'm using ESP32 Https Server on a WROOM-32 board.
I created a ResourceNode which serves an html page using GET method.
If I connect to that page using Chrome or Firefox from my PC it works without any problem.

The frontend for this project is managed by an Android application that connects to the ESP32 server through WiFi, using the WebView component.
Unfortunately, when the WebView component loads the page, the HTTP server hangs without logging anything.

If I turn on the debug log, I can see the headers coming in, but the "[HTTPS:D] Headers finished, FID=60" message doesn't show up and the HTTP server stalls in a not responding state.

After some debugging I found out a possible endless loop at this point:
https://github.com/fhessel/esp32_https_server/blob/daf6ca1c4359e9c9fdea561a748ebc8be029fb23/src/HTTPConnection.cpp#L306

It seems that if the last char is "\r" and the condition at line 295 is not met, then an endless while loop occurs because _bufferProcessed will not be incremented.

I simply incremented the _bufferProcessed variable after line 306 in order to fix my problem.
I just wanted to make you aware of this, just in case you want to fix it in the next releases.

Thanks and congratulations for this excellent project!

bulrog95, 2021-01-15:
Hello,
I can confirm the same bug found in a javascript redirect during page load, but increment doesn't work for me , i have added this to correct the server freeze:
[image]

Thanks @erosinnocenti for this issue, i have gain many time in this bug ,
and it's true esp32_https_server is an excellent https server!

fhessel, 2021-01-23:
Thank you both for your feedback. I'm still working on getting hardware integration tests running, I'll try to reproduce this issue in a test and then provide a fix for the next release.

vvigilante, 2021-04-20:
Hello, I believe this pull request is the solution to the bug: https://github.com/fhessel/esp32_https_server/pull/123

vvigilante, 2021-06-05:
@fhessel , Hey Frank! Shall we proceed with merging my pull request? Please tell me if I can help somehow.

Dilbert66, 2021-11-18:
I was getting consistent lockups accessing a page using the chrome Android browser due to this bug in the readline function.  The  increment fix took care of that issue.  I still get handshake errors from that browser though on initial connection so that's a different issue.  Desktop browsers such as IE and Chrome are fine.   Wow, this repository hasnt been updated in years?
Is there a new ESP SSL library that people are using?  There is not much out there that works.. This one at least accomplishes what  I need (which includes ssl websockets).

proddy, 2021-11-19:
> Is there a new ESP SSL library that people are using? There is not much out there that works.. This one at least accomplishes what I need (which includes ssl websockets).

I've been looking for one too for a long while now and was hoping to switch over to this library - but was put off by its inactivity. I'm still using ESPAsyncWebServer (non SSL) and although it has its problems, there's a lively community (although somewhat negative) and always someone that's managed to hack around its oddities.

KaloNK, 2021-11-29:
Same here, so i have decided to fork this library ( https://github.com/KaloNK/esp32_https_server ) and at least merge some pending pull requests.