fhessel/esp32_https_server #122: Example Websocket-Chat only seems to work with firefox browser

jaboh, 2021-04-19:
**Describe the bug**
The Websocket-Chat example does not seem to work with Chrome and Safari.

**How To Reproduce**
Steps to reproduce the behavior:
1. Add the following line to HTTPSServer.hpp:
`#define HTTPS_LOGLEVEL 4`
for a detailed log of the server's activity.
2. Compile and upload.
3. Reset the esp32 board.
4. Access the esp32 server using the Firefox browser from two browser windows.
5. Make the connection as two different users.
6. Observe that the example works nicely.
7. Repeat 4-5 from two Safari browser windows.
8. Note that the browser drops the socket connection immediately after serial monitor message "Creating new chat client!". Same behavior on Mac OS and IOS.
9. Repeat 4-5 from two Chrome browser windows.
10. Note that the server hangs: it never prints the "Creating new chat client!" message.

**Expected Behavior**
The (very nice) Websocket-Chat example is expected to work on most of the popular browsers.

**Actual Behavior**
The Websocket-Chat example works beautifully, but only with Firefox.

**ESP32 Module**
Specifications of my module:
- Board: AI Thinker ESP32-CAM
- RAM/PSRAM: 320K
- Flash Size: 4MB
- Separate UART USB adapter

Same behavior on Wemos D1 mini board, btw.

**Software (please complete the following information if applicable)**
 - Arduino 1.8.13
 - OS: Ubuntu 20.04 and Mac OS 11.2.3
 - Client used to access the server: Firefox 86.0.1 and 87.0, Chrome 89.0.4389.128, Safari 14.0.3 (16610.4.3.1.7)
 - No experience with curl

**Additional context**
Copies of the serial monitor are attached
[debug_firefox.txt](https://github.com/fhessel/esp32_https_server/files/6336263/debug_firefox.txt)
[debug_safari.txt](https://github.com/fhessel/esp32_https_server/files/6336265/debug_safari.txt)
[debug_chrome.txt](https://github.com/fhessel/esp32_https_server/files/6336267/debug_chrome.txt)

.

vvigilante, 2021-04-19:
Hello, I had a problem with chrome that exhibited the same symptoms. Oddly enough it worked much better with Edge (did not try with firefox or safari).
I solved with this pull request: https://github.com/fhessel/esp32_https_server/pull/123

Let me know if this solves your problem.

jaboh, 2021-04-19:
Thanks a lot Vincenzo, that solves the Chrome issue!
The Safari issue remains (client disconnects for some hard to detect cause), but with Chrome and Firefox both working this is now a minor issue as most iphone/ipad users are familiar with Chrome as well as Safari.

closed 2021-04-19 by jaboh