fhessel/esp32_https_server #127: Cannot find file hwcrypto/sha.h

wijnsema, 2021-05-28:
When I include the code from Self-Signed-Certificate.ino into my Platform IO project, the code will not compile.
It can not find the file hwcrypto/sha.h

I'm using Platform IO 5.1.1 with the newer framework-arduinoespressif32 platform:

This is my platform.ini

```
[env:esp32dev]
platform = espressif32
framework = arduino
board = esp32dev
board_build.partitions = partitions.csv 
build_flags = 
	-DARDUINO=250
	-DHW=4
	-DESP32
	;-DTESTBAR
	-w
upload_speed = 921600
monitor_speed = 115200
monitor_filters = esp32_exception_decoder, default
platform_packages = 
	framework-arduinoespressif32 @ https://github.com/marcovannoord/arduino-esp32.git#idf-release/v4.0
	toolchain-xtensa32 @ 3.80200.200512
lib_deps =
	ArduinoJson
	Adafruit BME280 Library
	ClosedCube STS35
	ClosedCube I2C Driver
	adafruit/Adafruit BusIO
	adafruit/Adafruit MCP23008 library
	PageBuilder
	esp32_https_server
```

Thanks!

641i130, 2021-07-09:
I'm also having this issue, though it might be related to a dependency...

641i130, 2021-07-09:
Actually, I figured it out by doing this:
https://github.com/espressif/esp-idf/issues/3245
Its related to something else I think. Just do that work around.

wijnsema, 2021-08-24:
Thanks Galileo, this indeed is working!

closed 2021-08-24 by wijnsema