fhessel/esp32_https_server #128: std::string versus String for response->print()

frankcohen, 2021-05-30:
Great job writing this library! It's helped me so much.

I wrote a handler as:

void handle_getpassword(HTTPRequest * req, HTTPResponse * res){
  ...
  std::string ssid;
  req->getParams()->getQueryParameter("ssid", ssid);  
  ...
  res->print( ssid );
  ...
}

I get a compile time error:

/Users/frankcohen/Documents/Arduino/EleksHack/EleksHack.ino: In function 'void handle_getpassword(httpsserver::HTTPRequest*, httpsserver::HTTPResponse*)':
EleksHack:497:20: error: no matching function for call to 'httpsserver::HTTPResponse::print(std::__cxx11::string&)'
   res->print( ssid );

Seems like there is no print method that takes the C++ std:string. It works with a String value.

What should I do?

-Frank

amandel, 2021-05-30:
try `res->print( ssid.c_str() );`

frankcohen, 2021-05-31:
That works. Thank you. -Frank

kinafu, 2021-07-05:
@frankcohen you might want to mark this issue as "closed" 😊

closed 2021-07-05 by frankcohen