fhessel/esp32_https_server #135: Guru meditation error on Self-Signed-Certificate example

foltos, 2021-07-15:
I created a project where I use the Self-Signed-Certificate example. After upload the code the device has Guru meditation error.
Can you help me what is the problem?

The error message:
Creating a new self-signed certificate.        
This may take up to a minute, so be patient ;-)
Creating the certificate was successful
Setting up WiFi
.....Guru Meditation Error: Core  panic'ed (Interrupt wdt timeout on CPU0)
Core 0 register dump:
PC               0x4014eb      PS        0x00060834    A0        0x8014e7a      A1       0x3fb890
A2               0x3fae8         A3        0x00000000    A4        0x02ac61        A5       0x00000000  
A6               0x3fafc          A7        0x00060023    A8        0xfd1ee          A9       0xfd1ee 
A10             0x02ac61      A11       0x00000000   A12       0x00000001   A13      0x00000001  
A14             0x00060820  A15       0x00000000   SAR       0x00000000  EXCCAUSE 0x00000005
EXCVADDR 0x00000000  LBEG     0x4000ce        LEND     0x4000cf       LCOUNT   0xffff  

ELF file SHA256: 000000000000000

Backtrace:0x4014eb:0x3fb8900x4014e77:0x3fb8b 0x400897e:0x3fb8d

ore 1 register dump:
PC               0x4018928e   PS        0x00060c4     A0          0x8014cc        A1             0x3fb6e  
A2               0x00000000   A3        0x80000001  A4          0x00000000    A5             0x00000001  
A6               0x00060a0     A7        0x00000000  A8          0x00060023    A9             0x3fb7d  
A10             0x00000001   A11      0x00000001  A12        0x800899f      A13            0x3fb5e  
A14             0x00000000   A15      0x3fb3e         SAR       0x00000000    EXCCAUSE 0x00000005
EXCVADDR 0x00000000   LBEG     0x00000000  LEND     0x00000000    LCOUNT    0x00000000  

ELF file SHA256: 000000000000000

Backtrace:0x4018928e0x3fb6e 0x4014cc:0x3fb700 0x4008a99:0x3fb720 0x400897e:0x3fb740

Rebooting...

ESP32 Module
- ESP-WROOM-32
- Flash Size: 4 MB

Software
 - IDE and Version: PlatformIO 5.5.1
 - OS: Windows 10

wladwnt, 2021-07-25:
I have the same issue:

0x40086fc4: _xt_lowint1 at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/xtensa_vectors.S line 1118
0x40086fc4: _xt_lowint1 at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/xtensa_vectors.S line 1118
0x4008cd1b: multi_heap_internal_unlock at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/multi_heap.c line 380
0x4008d37e: multi_heap_malloc at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/multi_heap_poisoning.c line 321
0x40081cc5: heap_caps_malloc at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/heap_caps.c line 232
0x40081ea1: heap_caps_calloc at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/heap_caps.c line 232
0x40081ec4: esp_mbedtls_mem_calloc at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/mbedtls/port/esp_mem.c line 25
0x4014bfc1: mbedtls_calloc at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/mbedtls/mbedtls/library/platform.c line 93
0x4014864e: mbedtls_mpi_grow at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/mbedtls/mbedtls/library/bignum.c line 2661
0x4014873b: mbedtls_mpi_copy at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/mbedtls/mbedtls/library/bignum.c line 2661
0x40149053: mbedtls_mpi_mul_mpi at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/mbedtls/mbedtls/library/bignum.c line 2661
0x40149117: mbedtls_mpi_mul_int at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/mbedtls/mbedtls/library/bignum.c line 2661
0x401494b8: mbedtls_mpi_div_mpi at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/mbedtls/mbedtls/library/bignum.c line 2661
0x40149657: mbedtls_mpi_mod_mpi at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/mbedtls/mbedtls/library/bignum.c line 2661
0x401498d6: mbedtls_mpi_exp_mod at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/mbedtls/mbedtls/library/bignum.c line 2661
0x401423eb: rsa_prepare_blinding at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/mbedtls/mbedtls/library/rsa.c line 425
0x40142d3f: mbedtls_rsa_private at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/mbedtls/mbedtls/library/rsa.c line 962
0x40143958: mbedtls_rsa_rsassa_pkcs1_v15_sign at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/mbedtls/mbedtls/library/rsa.c line 2108
0x401439eb: mbedtls_rsa_pkcs1_sign at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/mbedtls/mbedtls/library/rsa.c line 2151
0x4014c6ad: rsa_sign_wrap at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/mbedtls/mbedtls/library/pk_wrap.c line 145
0x401414d7: mbedtls_pk_sign_restartable at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/mbedtls/mbedtls/library/pk.c line 560
0x40141515: mbedtls_pk_sign at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/mbedtls/mbedtls/library/pk.c line 560
0x4014833d: mbedtls_x509write_crt_der at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/mbedtls/mbedtls/library/x509write_crt.c line 493
0x400f2bb9: _ZN11httpsserverL10cert_writeERNS_7SSLCertENSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEES7_S7_$constprop$18 at C:\Users\wlad\Documents\Arduino\libraries\ESP32_HTTPS_Server\src/SSLCert.cpp line 51
0x400f2dea: httpsserver::createSelfSignedCert(httpsserver::SSLCert&, httpsserver::SSLKeySize, std::__cxx11::basic_string  , std::allocator  >, std::__cxx11::basic_string  , std::allocator  >, std::__cxx11::basic_string  , std::allocator  >) at C:\Users\user1\Documents\Arduino\libraries\ESP32_HTTPS_Server\src/SSLCert.cpp line 297 (discriminator 4)

wladwnt, 2021-07-25:
OK, the same as in
[https://github.com/fhessel/esp32_https_server/issues/48#issue-483127431](https://github.com/fhessel/esp32_https_server/issues/48#issue-483127431)