fhessel/esp32_https_server #148: Remove deprecated openssl/ssl.h

petoknm, 2022-02-06:
I'm getting a warning about deprecated openssl component, so I guess it would be nice to use what Espressif recommends now, which is `esp_tls.h`.
I tried switching the headers and they provide different API so it doesn't compile right off the bat.
I wouldn't mind making a PR but I don't have much experience with crypto/TLS so I would need some guidance on the semantic differences between the two APIs.

```
#warning "OpenSSL component will be removed from ESP-IDF in v5.0, please use esp_tls instead"
```

georeb, 2024-04-28:
ESP-IDF v5.0 was released on Dec 2 2022, this library has still not been updated from OpenSSL, and this issue still exists. Any update @fhessel ?

al-TideLight, 2024-07-04:
I am also having this issue, has anyone found a work around?

petoknm, 2024-07-05:
Try out using this MR: https://github.com/fhessel/esp32_https_server/pull/169