fhessel/esp32_https_server #153: File upload in HTTPS

project-header, 2022-04-08:
Hello,

**Describe Your Goal**
I create this issue because I would like to upload files with esp32_https_server.

Before I was using WebServer.h which allowed to have the upload() function.
Then I switched to HTTPS with HTTPSServer.hpp and I was able to convert all my code (which is functional in HTTPS) except for the file upload.

Here is approximately the result I would like to obtain :
https://github.com/espressif/arduino-esp32/blob/master/libraries/ArduinoOTA/examples/OTAWebUpdater/OTAWebUpdater.ino

**What Does Your Project Look Like ?**
Currently, I use the following libraries
```c++
#include <HTTPClient.h>		/// Pour communiquer avec l'API
#include <SPIFFS.h>
#include <EEPROM.h>
#include <WiFi.h>
#include <Update.h>
#include <string>

#include <HTTPSServer.hpp>
#include <SSLCert.hpp>
#include <HTTPRequest.hpp>
#include <HTTPResponse.hpp>
#include <HTTPURLEncodedBodyParser.hpp>
```

**Module ESP32**
I use KitC v4 ESP32-WROOM-32
Hardware : ESP32 240MHz, 520KB RAM, 16MB Flash

**Software (please complete the following information if applicable)**
- IDE : Sublime Text v3.2.2 & PlateformIO v5.1.1
- OS : Windows 10 Famille v21H2
- Client used to access the server : Firefox mobile v98.3.0

**Additional context**
Thank you in advance for your help !

trullock, 2024-01-15:
This is my OTA Code which works

```
this->nodeUpdateFirmware = new ResourceNode("/firmware-update", "POST", [&](HTTPRequest* req, HTTPResponse* res) {
		HTTPMultipartBodyParser parser(req);
		
		while(parser.nextField()) {

			std::string name = parser.getFieldName();
			std::string filename = parser.getFieldFilename();
			std::string mimeType = parser.getFieldMimeType();
			
			Serial.printf("handleFormUpload: field name='%s', filename='%s', mimetype='%s'\n", name.c_str(), filename.c_str(), mimeType.c_str());

			if (name != "firmware") {
				Serial.println("Skipping unexpected field");
				break;
			}

			if (!Update.begin(UPDATE_SIZE_UNKNOWN, U_FLASH))
			{
				Serial.println("Error starting update");
				break;
			}

			while (!parser.endOfField())
			{
				byte buf[512];
				size_t readLength = parser.read(buf, 512);
				if(Update.write(buf, readLength) != readLength)
				{
					if (Update.hasError())
					{
						StreamString str;
						Update.printError(str);
						res->setHeader("Content-Type", "text/plain");
						res->setStatusCode(500);
						res->print(str);
					}
					Update.end();
				}
			}
			
			if(Update.end(true))
			{
				INFOln("Update success.");
				INFOln("    Rebooting...");

				res->setStatusCode(200);
				res->finalize();
				delay(100);
				ESP.restart();
			}
			else
			{
				if(Update.hasError())
				{
					StreamString str;
					Update.printError(str);
					res->setHeader("Content-Type", "text/plain");
					res->setStatusCode(500);
					res->print(str.c_str());
				}
				Update.end();
			}
		}
	});
```

adenrill, 2024-01-31:
Hi, thanks a lot for providing your code, is it possible to see the entire code, I don't understand some parts of it and I can't manage to make it work with my script, seeing all your code would be really helpful, thank you again

trullock, 2024-02-04:
> Hi, thanks a lot for providing your code @trullock, is it possible to see the entire code, I don't understand some parts of it and I can't manage to make it work with my script, seeing all your code would be really helpful, thank you again
> 
> For example, I get an error saying that HTTPMultipartBodyParser is not declared, is it coming from another library ?

Well yes, you have to include it:

`#include <HTTPMultipartBodyParser.hpp>`

It will be easier to help you if you post your problematic code. Make a Stack Overflow post or something and I can try and help you there