fhessel/esp32_https_server #156: Setting the Core Debug Level gives compile errors

PhotoKevin, 2022-07-07:
Using the Arduino 2.0 IDE, setting Tools->Core Debug Level to Debug gives compile errors. Levels None and Warn are ok. 
```

In file included from ...\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.4\cores\esp32/esp32-hal.h:76,
                 from ...\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.4\cores\esp32/Arduino.h:36,
                 from ...\Documents\Arduino\libraries\ESP32_HTTPS_Server\src\WebsocketHandler.hpp:4,
                 from ...\Documents\Arduino\libraries\ESP32_HTTPS_Server\src\WebsocketHandler.cpp:1:
...\Documents\Arduino\libraries\ESP32_HTTPS_Server\src\WebsocketHandler.cpp: In function 'void httpsserver::dumpFrame(httpsserver::WebsocketFrame)':
...\Documents\Arduino\libraries\ESP32_HTTPS_Server\src\WebsocketHandler.cpp:20:5: error: 'TAG' was not declared in this scope
     TAG,
     ^~~
...\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.4\cores\esp32/esp32-hal-log.h:127:72: note: in definition of macro 'log_i'
 #define log_i(format, ...) log_printf(ARDUHAL_LOG_FORMAT(I, format), ##__VA_ARGS__)
                                                                        ^~~~~~~~~~~
...\Documents\Arduino\libraries\ESP32_HTTPS_Server\src\WebsocketHandler.cpp:19:3: note: in expansion of macro 'ESP_LOGI'
   ESP_LOGI(
   ^~~~~~~~

exit status 1
```

PhotoKevin, 2022-10-24:
#160 says adding esp_log.h doesn't fix it. I hope it's maintained, I want https.

WayneKeenan, 2023-08-17:
In `esp32_https_server/src/WebsocketHandler.cpp` change:   
```
ESP_LOGI(
         TAG,
```

to:

```
HTTPS_LOGI(
```