fhessel/esp32_https_server #160: Is this library still maintained?

nicklasb, 2022-09-04:
Hi everybody,

I will try and use this library, but looking around it seems slightly abandoned.
Has people moved on to other libraries or am I mistaken?

Kind regards, Nicklas

MYBLtd, 2022-10-13:
Don't know, but I tried today and had some problems:
- in HTTPConnection.hpp had to change #include <hwcrypto/sha.h> to #include <esp32/sha.h>
- now I have this error: Arduino/libraries/ESP32_HTTPS_Server/src/WebsocketHandler.cpp:20:5: error: 'TAG' was not declared in this scope

squonk11, 2022-10-13:
For the error in WebsocketHandler.cpp you need to include another header. Currently I don't know which one because I don't have access to my PC. But this question has already been anwered some time ago.

squonk11, 2022-10-13:
Look here:
[link](https://github.com/fhessel/esp32_https_server/issues/156#issuecomment-1179163883)

nicklasb, 2022-10-13:
Personally, partially due to the lack of activity, I went on to use the esp_http_server.h instead, worked OK.
This thread seems to have been diverted slightly, but on my part I can close this issue.

MYBLtd, 2022-10-13:
I did some testing with debug level and including esp_log.h
Setting debug level back to warn resolves the TAG error.
Any more verbose will generate the error.
Including esp_log.h makes no difference.

WayneKeenan, 2023-08-17:
A monkeypatch fix for the logging error is in my #156 comment