fhessel/esp32_https_server #163: hwcrypto/sha.h

podaen, 2022-10-04:
The #include <hwcrypto/sha.h> could not be found with the latest arduino esp32 (2.0.4).

Pablo2048, 2022-10-04:
https://github.com/fhessel/esp32_https_server/issues/143

https://github.com/fhessel/esp32_https_server/issues/136

ad some PR...

podaen, 2022-10-04:
I'm new to this and just checking how this thing works. I have some add warning for the next version of IDF.

> ssl.h: 8:2: warning: #warning "OpenSSL component will be removed from ESP-IDF in v5.0, please use esp_tls instead" [-Wcpp]
>    #warning "OpenSSL component will be removed from ESP-IDF in v5.0, please use esp_tls instead"
>    ^~~~~~~

Pablo2048, 2022-10-04:
https://github.com/fhessel/esp32_https_server/issues/148