fhessel/esp32_https_server #168: ESP32 With W5500 Ethernet module not working for https webserver

Harishferose, 2022-12-20:
I am trying to access the HTTPS server through ethernet, to achieve this I have modified the HTTPS (Async Server Example) code to handle an ethernet connection (by adding an ethernet connection code).

Having both wifi and ethernet connected the code compiles and uploaded successfully in this case with WIFI IP I can able to access the web server but can't access it with EthernetIP 

Next step I removed the WIFI connection code alone and Checked with the code, It compiles successfully but when the code is uploaded ESP32 board keeps on rebooting.

I want to achieve an HTTPS web server through the ethernet connection in this board.

**ESP32 Module**
- Adafruit feather espressif esp32 s3 mini

sdrshnptl, 2022-12-21:
Hi @Harishferose 
Please share code snippets to investigate in detail.

Harishferose, 2022-12-21:
Hi @sdrshnptl, Please find the code in the below attachment, Please have a look at it. Thanks in advance

[wifi.txt](https://github.com/fhessel/esp32_https_server/files/10276647/wifi.txt)