fhessel/esp32_https_server #177: Crash when fast reloading web-site

trlink, 2023-12-04:
Hi all, 

I ran into a problem, when I had a client fast refreshing the loading of the web-site hosted on a ESP:

The connection was opened and then the client closed the connection, which was written into the log, but in the void HTTPConnection::loop(), this was not checked, the request still was answerred:

So i changed the before the switch to:

if (!isError() && _clientState != CSTATE_CLOSED) 

Which solved the crashes...

trullock, 2024-01-16:
@trlink I see you've got an active fork of this, can I help contribute? we seem to be in about the same place bug-finding wise. Your repo doesn't allow issues directly

trlink, 2024-01-16:
Hey Andrew,Sure, I need to check how to enable it...--Diese Nachricht wurde von meinem Android Mobiltelefon mit GMX Mail gesendet.