fhessel/esp32_https_server #183: Not supported for ESP32-C6 and other introduced in V3+ Arduino framework

virco, 2024-06-29:
arduino-esp32 Library introduces support of ESp32-C6 and other new SoC's.

Unfortunately, it looks like "openssl/ssl.h" is abandoned and the recommendation is to use esp_tls.h instead.

al-TideLight, 2024-07-04:
Im also having this issue, would love to know if you figured out a sucessful work around for this.

virco, 2024-07-04:
> Im also having this issue, would love to know if you figured out a sucessful work around for this.

No, still waiting for for some solution or workaround to be shared :-(

AVDeveloppement, 2024-08-31:
Hello, have you changed for another like https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/protocols/esp_https_server.html ?
I think to do it since i need update some of my work to last library :(
Have you a fix or another https library ?
Thanks.